Polidea/RxBluetoothKit #13: Setting up notification on characteristic - possible race condition.

Kacper20, 2016-04-19:
Hi,

So basically: when we're setting up notification on characteristic that's proper way to do it(without use of convenience functions):
```swift
characteristic.setNotifyValue(true)
    .flatMap { $0.monitorValueUpdate() }
    .subscribeNext {
        let newValue = $0.value
    }
```
So, first notification set is called, and after that .Next with characteristic that notification status was updated is emitted. Monitor on updates is called after that. The result is that in between of these two calls update event could happen and it'll be lost.

Possible way to prevent this behavior is to do it like this:
```swift
Observable.of(characteristic.setNotifyValue(true).ignoreElements(), characteristic.monitorValueUpdate())
.subscribeNext {
//Read value
}
```
But i'm worried it's too verbose to user.
So maybe we should leave it like it is(so basic two functionalities) and add to this function like `setupNotificationAndMonitorUpdates`, that'll work for setting notification in state ON.
We could implement this like i showed in second snippet.
What do you think?

literator, 2016-04-20:
I think that it would be good to have `setNotifyValueAndMonitorUpdates(values: true)`.
This kind of races are not easy to understand at first, especially if someone is new to the subject.

Kacper20, 2016-04-21:
Eliminated race conditions and created this method. :-)))

closed 2016-04-21 by Kacper20