Polidea/RxBluetoothKit #15: Scan sharing

iamszabo, 2016-04-27:
I've just switched to the new 1.0.0 version of the RxBluetoothKit. So far it's amazing and huge thanks for it. But upgrading from 0.3.* I faced with a problem that's worked before the new version.

Based on the documentation it can share the already scanned devices if the identifiers of new scans are subsets for both scans. I used to scan with nil, but also tried with one of the advertised CBUUID. 
I make the same call for both peripherals but no luck with the scan sharing. I can only make it work by terminating the scan after I found what I'm looking for.
```
    manager.monitorState()
      .filter { state in state == .PoweredOn }
      .take(1)
      .flatMap { _ in self.manager.scanForPeripherals([Constants.DeviceInformationService], options: nil) }
      .filter { peripheral in peripheral.peripheral.name == name }
      .take(1)
      .flatMap { $0.peripheral.connect() }
```

Kacper20, 2016-04-27:
Hi,
If i understand you correctly:
First you're making scan with `nil`, and when it's still active you're doing next scan with specified identifier, right?
You're talking about terminating scan - which scan is terminated? Could you provide code with both scans, so that we can understand the whole context?
Thanks for using library.

iamszabo, 2016-04-27:
Hi @Kacper20,

I was just saying I tried both variations, scanning with nil and scanning with specified CBUUIDs. Neither of them result with shared scans for me.
My point is I would like to make the usage of sharing scans (scanning concurrently) and I don't want to depend on the order of scanning queue (but this was the only thing that I could achieve with this version).
Maybe a bigger snippet of my code will help you understand what I would like to achieve.
```
 func connect(name: String) -> Observable<Peripheral> {
    return manager.monitorState()
      .filter { state in state == .PoweredOn }
      .take(1)
      .doOnNext { _ in print(name) }
      .flatMap { _ in self.manager.scanForPeripherals(nil, options: nil) }
      .filter { peripheral in peripheral.peripheral.name == name }
      .take(1)
      .flatMap { $0.peripheral.connect() }
  }

func connectPeripherals() {
   let left = connectivityService.connect("left")
   let right = connectivityService.connect("right")
   Observable.zip(left, right, resultSelector: { (leftConnected, rightConnected) in return (leftConnected, rightConnected) })
   .subscribeNext { _ in print("both connected  now you can watch 🐱 videos") }
}
```

Kacper20, 2016-04-27:
Hello,
We understand the problem. We've potentially fixed it, and added some more tests. But we want to check your snippet on the device to be sure. If everything will go smoothly, 1.01 will be released later today.

Kacper20, 2016-04-27:
Hi @iamszabo 
We believe that this should be fixed in `1.0.1`. We've tested this in scenario like yours.
Please check if it works now, and let us know :)

iamszabo, 2016-04-27:
@Kacper20  awesome, it's working now. Thanks for the quick fix and keep up the good work ;)

closed 2016-04-27 by Kacper20