Polidea/RxBluetoothKit #21: Peripheral disconnection

iamszabo, 2016-05-04:
Hey there,

Is there any way with the current version to know if a peripheral is disconnected?
If not are you planning to have an Observable that emits when the peripheral is disconnected in the near future?

Kacper20, 2016-05-04:
Hi,
Basically you could check it through `state` property on Peripheral. If you want to observe on it - in current version there's no way to do it.
If you'll receive disconnection while doing any action like discovering services, characteristics, etc. `Observable.error(BluetoothError.PeripheralDisconnected)` will be emitted, so you could watch on this particular error.

Besides that - i think that we could implement this, because it suits very well in our set of monitor methods.
So the answer is: yes, we could implement it in near future, but first we have to think a bit on API. 
I think that the most convenient  would be to have method `monitorConnectionState` that will observe on `didConnect` and `didDisconnect` and emit enums with values `Connected` and `Disconnected`.

Out of curiosity: in which case do you need this method? Maybe context will help us to design better interfaces. :-) 

Thanks,

iamszabo, 2016-05-04:
I cannot think any better than having a `monitorConnectionState` on the peripheral as you suggested.
Cause it can be important to know which peripheral is disconnected and act according to that if multiple devices are connected.
But in this case I think it's needed to share the given peripheral for multiple observable sequences.

This fits to my regular use case cause I subscribe for multiple services/characteristics at the same time.

So I would say, let's do that 👍

Kacper20, 2016-05-04:
Ok, we thought about it internally, and we have a better way to do this. We'll use `Variable` to monitor state changes of `Peripheral`. This will emit `CBPeripheralState`, so you'll have access to more detailed data about state. To give consistent API, we will mark current monitoring methods on `BluetoothManager` state as deprecated, and implement new APIs there too. 
I think that it'll not be implemented today, rather tomorrow, because it requires some thought to do it properly & not break the API.

Thanks for feedback, really appreciate it. :-)

Kacper20, 2016-05-06:
Hi,
We have code ready to merge, but still need to make some readme changes. 
I'll try to push it tomorrow to the branch master and make a release.

Kacper20, 2016-05-06:
Should be fixed by #22 .
Take a look at it and let us know what you think.

iamszabo, 2016-05-06:
Wow! Awesome and quick work again. Thank you 👍

Kacper20, 2016-05-11:
Hi, is it working for you? We've discovered some issues, and we don't know if it happens to every device, or just a few. 
If it's not - don't worry - we'll deploy next version with quick fix.

iamszabo, 2016-05-11:
Hi. Yes it's working for me. I use the rx_state only for peripheral and I haven't touched my old code with monitorState() for the central.

Kacper20, 2016-05-11:
Thanks for answer, we'll dig into it.

closed 2016-05-11 by Kacper20