Polidea/RxBluetoothKit #24: BluetoothManager.rx_state property not working

mhle, 2016-05-09:
I switched to the new BluetoothManager API with version 1.1.0 using rx_state property instead of monitorState() but it seems I'm not getting state changes anymore. 

Running on iPad Mini 2 WiFi, iOS 9.3.1

For now I'm sticking to the old API which works fine.

Kacper20, 2016-05-09:
Hey there,

Thanks for telling us - we'll take a look at it today.

Kacper20, 2016-05-09:
This is a bug - we'll fix it shortly. For now please use `monitorState` function.

mhle, 2016-05-09:
Thanks for looking into it and for creating this awesome lib

Kacper20, 2016-05-12:
Hi,
It's fixed in 1.1.1 version. 
We'll investigate `rx_state` on Peripheral too, but it looks like it's working. We'll test it and give further info about it.

Thanks for reporting issue and have a great day :-)

mhle, 2016-05-16:
Hi I can confirm that it's working in 1.1.1. great work!

closed 2016-05-16 by mhle