Polidea/RxBluetoothKit #25: Support for state restoration and preservation

mane87, 2016-05-17:
Hi,
is state restoration and preservation supported? I could not find anything in the docs.
As far as i can tell from the sourcecode, it is not implemented

Is there any plan on implementing it?

Thanks for the awesome work so far 💯 👍

literator, 2016-05-17:
Hi,

Thanks for the feedback! We plan to dig in more into state restoration &
preservation topics and support for them.

Do you have particular needs or nice-to-haves regarding this API?

Cheers,

mane87, 2016-05-18:
Hi,
thanks for the fast answer!

So, i am currently building a prototype app for our startup:
http://timeular.com
(Please have look and you should get the idea pretty fast ;) )

Obviously, my app should be notified about changes for the up-facing-side (we use an accelerometer) of my accessory. Thats why i want to use state restoration and preservation.

If you could guide me i would also love to help you to implement it.

Cheers, Manuel

mhle, 2016-05-20:
I would love to see this feature as well! is there a workaround we can use until it's done? I.e. do we have access to the underlying corebluetooth functionality? Our app interacts with multiple BLE devices and needs to know when they're switched on and off so continuously needs to scan.

Cheers, 

Michael

Kacper20, 2016-05-20:
Hi,
Are you talking about `willRestoreState` from `CBCentralManager`?
It's not visible yet, but we can do this fairly quickly. I have never worked with state restoration. Do we need to do anything more? 
From our side - I'll  do research over the weekend, and will talk to you back after that. 

Thanks for using library and have a great weekend!

Kacper20, 2016-05-22:
Hi,
I've looked at it and it looks like it should be pretty easy to implement. Will do it probably in monday / tuesday. Once we'll test it it'll be released :)

Thanks for feedback

mane87, 2016-05-22:
Hi,
thanks for the awesome work!!
Once is ready i am going to test it and provide feedback.

Thanks again for the awesome work you are doing 💯 👍

mhle, 2016-05-22:
Thanks for quick reaction, looking forward to testing this new feature out! 👍

Kacper20, 2016-05-30:
Hi,
Today we've pushed 1.2 version, which contains requested features.
Please check it out and let us know about your impressions. 

Regards

closed 2016-05-31 by Kacper20

mhle, 2016-05-31:
Super, thanks for acting so fast!

Cheers, 

Michael