Polidea/RxBluetoothKit #26: Issue with running the Example

c0ng7u, 2016-05-23:
First of all, thanks so much for building an awesome bluetooth library, at least for a newbie like me. I was trying to run the Example project and ran into some errors. Could you please point me to the right direction? 

Thank you very much.

(screenshot attached)

[image]

Kacper20, 2016-05-23:
Hi,
Which version of a lib your're using?

c0ng7u, 2016-05-23:
Thanks for your quick response. It's RxBluetoothKit (0.3.2)

Kacper20, 2016-05-23:
Hi,
Please update to the newest 1.1.1 version and let me know if everything works fine.

c0ng7u, 2016-05-23:
version 1.1.1 did the trick. In the podfile, if I used just 
pod 'RxBluetoothKit' 
it would keep downloading the 0.3.2 version. I had to use
pod 'RxBluetoothKit', :git => 'https://github.com/Polidea/
in order to get the 1.1.1

Thank you for you help

Kacper20, 2016-05-23:
Thats weird. It should download it automatically. But I'm glad that it works now. If you have any questions feel free to ask - we're happy to help. 
Regards

closed 2016-05-23 by Kacper20