Polidea/RxBluetoothKit #28: Ambiguous use of from

AndrewSB, 2016-06-14:
The library is great! Having one issue when I drag the `RxBluetooth` source code directly into my project. I get an error on every use of `Observable.from`, saying that use of `from` is ambiguous.

Any idea why this may be happening or how I can debug this?

AndrewSB, 2016-06-14:
Works great when I use Carthage

Kacper20, 2016-06-14:
Hi there,
We have never tried to integrate it directly, so it may not work as it should at the moment. We'll check it tomorrow and let you know about the solution right away .

Thank you very much for feedback,

Kacper20, 2016-07-04:
Hi,

Could you include sample project here? It would be very helpful. :-)

literator, 2016-08-02:
@AndrewSB @Kacper20 any update here?

closed 2016-09-13 by Kacper20