Polidea/RxBluetoothKit #29: Thoughts on adding CBPeripheralManager to the kit?

AndrewSB, 2016-06-14:
I was going to tackle this in a PR - any suggestions/ insight before I start?

AndrewSB, 2016-06-14:
Would it make sense to move all the CentralManager code into a sub-object? Something like Bluetooth.CentralManager & Bluetooth.PeripheralManager?

Kacper20, 2016-06-14:
That's great!
My only suggestion is to try to follow our conventions to delegates to be consistent across the whole library.
About naming - I would advise to first try to implement it separately, as a non breaking addition to the lib - maybe some object called `BluetoothPeripheralManager`. Later we could think about naming. We don't want to make big changes here(to the existing codebase), so that users of the lib would not have to worry about big breaking changes. 
I'm really looking forward to your contribution. If you have any questions - feel free to let us know and we'll be happy to discuss them.

AndrewSB, 2016-06-14:
will do! Thanks @Kacper20.

Created #30 to track my progress & for discussion

closed 2016-06-14 by AndrewSB