Polidea/RxBluetoothKit #32: replay(1)?

AndrewSB, 2016-06-15:
Apologies for my unfamiliarity with rx operators, but can the use of `.replay(1)` on a stream be explained? 

From my use I believe it's convenient way to multicast something? So multiple subscribers get the same underlying sequence?

`.replay(1)` from my understanding would then make it so both the first and second subscriber of the observable would be subscribed to the same underlying stream.

Here for example: https://github.com/Polidea/RxBluetoothKit/blob/master/Source/RxCBCentralManager.swift#L127
What does that accomplish?

literator, 2016-07-07:
Motivation for using `replay(1)` for `rx_state` is to give you very last state when you subscribe to this observable.

Kacper20, 2016-07-13:
Closing issue for now - if you'll have any additional questions please feel free to reopen :)

closed 2016-07-13 by Kacper20