Polidea/RxBluetoothKit #33: Working with BLE in background

xtazz, 2016-06-17:
Hi! I'm going to use your library as a replacement of my custom implementation in the current project and just want to know if you handle app background/foreground transitions properly as described in the Apple docs: 
>Reinstantiate Your Central and Peripheral Managers
>When your app is relaunched into the background by the system, the first thing you need to do is >reinstantiate the appropriate central and peripheral managers with the same restoration identifiers as >they had when they were first created.

https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html

Thanks for your efforts!

Kacper20, 2016-06-17:
Hello,
Yes, you should just reinstantiate `BluetoothManager` with proper identifier (the one you gave to it previously), and immediately subscribe to `listenOnRestoredState()` method. If system has any state restored, you'll receive it packed in `RestoredState` struct. In case of any problems just let us know about it, and we'll try to help you.

Have a great day

xtazz, 2016-06-17:
Thanks for your answer. So I'll loose all my subscriptions and will have to resubscribe to the new manager? If you do abstraction layer above central manager maybe it's better to reinstantiate it automatically inside your lib to save all subscriptions to make easy and smooth transition from/to background? what do you think?

Kacper20, 2016-06-17:
When you'll have in background - there's no problem. When your app comes to background, all of the subscribtions are valid, and when for example next peripheral is discovered - your app just does the work it's intended to do(code is executed).
Bluetooth state restoration is useful, when your app was killed(by user of system) and you want to receive peripherals connected during the time, when your app was not running in the system(not just foreground). In this case you have to use api I've provided. And i don't see any way we could improve it - in a lib we don't know at the fresh start of the app which subscribtions you've had.

If you have idea - i'm eager to listen and discuss it.

Kacper20, 2016-06-21:
Hi,
Is everything clear, or you still have some questions?
In case if something is not clear - i'm here to help :)

xtazz, 2016-06-21:
Hi! I'm preparing a test app with RxBluetoothKit to confirm that it works better than my custom implementation. I'll let you know if I'll have any additional questions. Thanks for help!

Kacper20, 2016-06-21:
Great. I'm closing it for now. If you have question - please reopen it :)

closed 2016-06-21 by Kacper20