Polidea/RxBluetoothKit #38: listenOnRestoredState() should be Public

mane87, 2016-06-27:
Hi,
finaly i am going to give staterestoration a try.

In you readme you write:
 **Later, just make sure to subscribe to listenOnRestoredState observable, and inspect RestoredState instance, which consists any useful info about restored state.**

But listenOnRestoredState() is private. Shouldn't it be public?

Cheers

Kacper20, 2016-06-27:
Hi,
Yea, that's right. That's our mistake. Will make it public in tomorrow's release. 
Thanks!

mane87, 2016-06-27:
Oh, cool!
Wasn't sure if I am overlooking something, otherwise I would have made a pull request.
Thank you very much for your fast answer!

mane87, 2016-06-29:
Hi,
any news for the update?

Should i make a pull request for the change?

Thanks again

Kacper20, 2016-06-29:
Hi there,

Now the version with fix is available on Github. It's 1.2.1.
Please wait a bit for availability in Cocoapods(pushed to trunk already).

Have a nice day and let us know in case of any problems :-) Thanks!

mane87, 2016-06-29:
Thanks.
I'll give it a try and let you know if there are any problems.

You guys are awesome!!! 👍🏼👍🏼👍🏼

Kacper20, 2016-06-29:
Closing for now. If anything will come up - just reopen it or(the better) make another issue.

closed 2016-06-29 by Kacper20