Polidea/RxBluetoothKit #39: monitorPeripheralDisconnection(peripheral: Peripheral) never gets called

mane87, 2016-07-02:
Hi,
i remove the battery from a connected peripheral, but *monitorPeripheralDisconnection* never emits a new value.

```swift
manager
            .monitorPeripheralDisconnection(peripheral)
            .debug()
            .subscribeNext{
                print("disconnected from: \($0)")
            }.addDisposableTo(disposeBag)
``` 

Am i doing something wrong?

Kacper20, 2016-07-04:
Hi there,

I'm looking into it right now. :-)

Kacper20, 2016-07-04:
Hi there,
I was trying to do this and it succeeded with our device. Today or tomorrow we will release another version - in example you'll have sample code that does this thing.

Also, I've added convenience method to the peripheral, to be able to call it more efficiently. 
If problem still occurs - please provide us some additional info about it :)

mane87, 2016-07-05:
Hi,
cool. I'll give it a try as soon as its available!

Thanks again for your awesome work ;)

closed 2016-07-05 by Kacper20