Polidea/RxBluetoothKit #41: RXBluetoothKit get notified on peripheral disconnect

JanBrinker, 2016-07-08:
Hi,
I'm currently trying out the RxBluetoothKit for a project and I want my app to connect to several peripherals simultaneously. Now I'd like to be notified in some way if one peripheral drops out in some way or closes the connection.

I could actively loop over the list of connected peripherals periodically and read out the **isConnected** property, but this seems a bit unclean to me. Is there a more elegant way to achieve this and get notified directly?

Thanks in advance :)

literator, 2016-07-08:
Hi,

Did you try subscribing to rx_state subject property on Peripheral?

Maciej Oczko
Polidea | Head of Engineering

JanBrinker, 2016-07-08:
Will try that and tell you how it went or if I need some more info, I still need to wrap my head around Rx.
But thanks for the tip :)

Kacper20, 2016-07-10:
Hello,

You could also try to subscribe to `monitorPeripheralDisconnection` on `BluetoothManager` method. All you need to do is to have reference to `Peripheral` instance.
We have this exact thing in our `Example` app. Maybe it's worth for you to take a look :-)

Kacper20, 2016-07-13:
Closing issue for now - if you'll have any additional questions please feel free to reopen :)

closed 2016-07-13 by Kacper20