Polidea/RxBluetoothKit #42: Access to underlying CBPeripheral

mobiletoly, 2016-08-08:
Let's say I already have an instance of RxBlueoothKit::Peripheral class available. How to get an access to underlying CBPeripheral instance? I have third-party library that performs Device Firmware Update (DFU) and it asks for reference to CBCentralManager and CBPeripheral, so would be great if I could just get it from already existing instance of RxBlueoothKit::Peripheral.

Thank you, your library is outstanding.

Kacper20, 2016-08-09:
Hi there,
Basically, you have reference to `RxBluetoothKit::Peripheral`. This instance has access to the `peripheral` property which is `RxPeripheralType` instance. This is a protocol, which allows us to test our implementation. We've built on it Fake objects, aswell as real `RxCBPeripheral` object. This object has property `let peripheral: CBPeripheral`. Unfortunately, because thanks to this `RxPeripheralType` protocol, usage assumes that you shouldn't access `RxCBPeripheral` because it's implementation detail. It's class with internal access so it's not really possible as of now.

One solution that comes to my mind quickly is to fork library and add `public` scope to two places:
```
public class RxCBPeripheral: RxPeripheralType {

	public let peripheral: CBPeripheral
	[...]
}`
Then, assuming that your `Peripheral` instance is named `peripheral` you'd write it like:
`let cbPeripheral = (peripheral.peripheral as! RxCBPeripheral).peripheral`

Which looks ugly, but should work properly.
We'll talk about your problem internally, because it's interesting case and we should take such into consideration. 
We'll let you know soon about our thoughts. 

Thanks for help in improving library!

Kacper20, 2016-08-26:
Hi,

Have you managed to deal with this problem somehow? 

Thanks!

jwhitley, 2016-11-23:
Hi all, I work with @mobiletoly now, and have gone through a few iterations of adding accessors for `CBCentralManager` and `CBPeripheral` to support our integration with Nordic's firmware update library.  Please see [this branch in our fork of RxBluetoothKit](https://github.com/LoopDevices/RxBluetoothKit/tree/jwhitley/cb-access) for the current work, and [its diff vs. master](https://github.com/Polidea/RxBluetoothKit/compare/master...LoopDevices:jwhitley/cb-access).

A significant challenge is providing access while avoiding a cascade of `public` properties and types within the library.  After a few iterations, the above has largely settled out.  The basic idea is to provide explicit accessors on `Peripheral` that let us obtain the wrapped CoreBluetooth objects.

FWIW, I view this as a starting point for discussion, not a definitive solution.  All feedback welcome!

Kacper20, 2016-11-24:
Hi there,

We're planning development of 4.0 version of library which will consist major changes in API and internal structure design of objects in lib. We think of access 
We want to achieve the same goal as you do, but without performing force casting, and rather to use swift protocols with associated types and be sure in compile time that the `CoreBluetooth` instance is there. You're welcome to take a look at the general idea in #79, you're feedback is welcome

Cheers,

tcurdt, 2017-08-02:
Bummer - I in the same boat. No access to the underlying structures is a deal breaker :-(
Is there any progress on this? or the 4.0 version?

Kacper20, 2017-08-02:
Hi @tcurdt,

We're working on 4.0, but we're partially blocked by the Swift lang functionalities. I will talk about this with my coworkers and try to develop temporary solution soon.

Thanks for patience,

tcurdt, 2017-08-02:
@Kacper20 Thanks for the quick reply!

Could you elaborate on the blocking? Or are you planning to switch to Swift4?

If the API changes in 4.0 anyway - maybe merging from @jwhitley and making a new release could be an option?

Kacper20, 2017-08-03:
Hi,

Yeah, of course. We're having entire new architecture for 4.0 that is somewhat similar in terms of interfaces but testing of implementation is based solely on protocols with associated types. It allows us to expose `CBPeripheral` object in a clean way, without any force casts in implementation classes.

It'll be only Swift 4.X, we'll be still maintaining the earlier versions of course. `X` is because our architecture right now is crashing the compiler when WMO is disabled, you can find the bug reported here: https://bugs.swift.org/browse/SR-5286. We're not sure when it'll be fixed.

Because right now it's impossible to access `CBPeripheral` and we don't want to keep it longer that way I think that we'll be implementing ugly forcecast to expose `CBPeripheral`. It's not ideal in terms of clean code, but works always and should be really useful for the users. I still have to talk with @Cierpliwy on this topic and we'll decide what path to take. It would be in `3.1` release of our lib.

Thanks,

Kacper20, 2017-08-04:
@tcurdt We've decided to go with this and #129 will add this functionality. Should be merged soon to master.

Thanks for patience

Kacper20, 2017-08-23:
It's now ready in 3.1 version of the library. Will be available in any future versions with more pure implementation.

closed 2017-08-23 by Kacper20