Polidea/RxBluetoothKit #43: WriteValue To Characteristic

marcelogfsouza, 2016-08-17:
Hi,
I'm trying to write some data into a BLE device, but haven't been able to succeed yet.
```swift
struct myStruct {
    private var connection: Observable<Peripheral>
    init(device: Peripheral) {
        self.connection = device.connect()
     }
	func writeCharacteristicFromDevice(data: NSData) -> Observable<Characteristic> {
    	       return connection.flatMap { $0.writeValue(data, forCharacteristicWithIdentifier: DeviceCharacteristics.Write, type: .WithoutResponse) }
	}
}
```
this is where I'm writing the value, then somewhere else I call and subscribe to it:
```swift
		contentView.sendDataButton.rx_tap
			.subscribeNext { [weak self] in
				guard let uSelf = self else {
					return
				}
                    uSelf.serviceHelper.writeCharacteristicFromDevice(data).subscribe(onNext: { (characteristic) in
						print("next")
						}, onError: { (error) in
						print("error")
						}, onCompleted: {
						print("completed")
						}, onDisposed: {
						print("disposed")
					}).addDisposableTo(uSelf.disposeBag)
				}
		}.addDisposableTo(disposeBag)
```
What happens here is: next is printed, then completed, then disposed.
If I add .WithResponse: error is printed (code=3 "writing is not permitted."), then disposed
The device's status is being monitored, it's possible to verify a successful connection, but not any writing attempt. It's as if the data is never arriving to the device.

The strange part is: the same thing was done in android with the same device and it works.
It was done using your android library `RxAndroidBle`. code:
```java
connectionObservable.flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(BLE_WRITE_CHARACTERISTIC_UUID,
                                                                                          new byte[]{
                                                                                            (byte) 0xaa,
                                                                                            (byte) 0x04,
                                                                                            (byte) 0x0C,
                                                                                            (byte) 0x00,
                                                                                            (byte) 0xbb
                                                                                          }))
                          .observeOn(AndroidSchedulers.mainThread())
                          .subscribe(bytes -> {
                            onWriteSuccess();
                          }, this::onWriteFailure);
```
Any clues on what might be wrong? I've tried different things but nothing worked so far.

Kacper20, 2016-08-18:
Hi there,
One thing i could think about is to check  `properties` property on `Characteristic`. So while handling `onNext` in piece of code you provided, check returned characteristic properties and let us know if it has `WriteWithoutResponse` and/or `Write` property. It looks like characteristic you've received is not writeable, so let's make sure it is. 
Let us know about the result :-) 

Thanks!

marcelogfsouza, 2016-08-18:
Hi there @Kacper20 ,
The characteristic properties is of type 'CBCharacteristicProperties.WriteWithoutResponse'.
Makes sense for the error when asking a response, but still have the problem on not writing in the device.
Any ideas?

Kacper20, 2016-08-19:
Hm. I'm thinking about it.
What we're doing is basically checking that you're Bluetooth state is all right - so that `Bluetooth` is on and device you're trying to reach is connected.
Later, call to the internal `CoreBluetooth` implementation is made. 
Could you place a breakpoint on the `self.peripheral.writeValue(data, forCharacteristic: characteristic.characteristic, type: type)` line in `Peripheral.swift` file? It's 286 line. Just to make sure that call is being made. If it's called - everything should work just fine. If doesn't - it's a problem and we'll work on fixes as soon as possible

Thanks,

marcelogfsouza, 2016-08-19:
So, I added a breakpoint and it stopped there. So it means that _self.peripheral.writeValue(data, forCharacteristic: characteristic.characteristic, type: type)_ being called. 
We use a tool to debug the peripheral that write on a console changes on the peripheral (if something connect, disconnect, write on a characteristic, etc). On our android app, using _RxAndroidBle_, it shows perfectly.  When the connection starts, finish and when there is a write on a characteristic. But on the iOS version, it only shows the connection start and finish, meaning that the peripheral is not receiving anything.
So you think it's a bug?

Thanks

Kacper20, 2016-08-19:
Hm, that's the only characteristic you're using? Is it readable? Are you able to read value of it, or value of any other characteristic? 
We could reject the possibility of using wrong characteristic - `Characteristic` object is created and works well. Also, write call to the underlying `CoreBluetooth` implementation is done, so as of now i can't think about possible problem from our side, because there's no other point of failure. 

Could you check this scenario with other characteristic, and/or check how read works? Maybe it's some kind of incompatibility with `CoreBluetooth`.

marcelogfsouza, 2016-08-19:
We're using 2 characteristics here. One to write and another to listen to notifications.
They're not readable. it's a bit of a different flow but are the devices specifications we got. 
To read a value we have to write a 'read' command and listen to the notify on the second characteristic.
I tried to check the properties of the others characteristic of the service (there are 5). 2 of them are **WriteWithoutResponse** and 3 of them I got **CBCharacteristicProperties(rawValue: 18)**.  I don't know exactly what it means.
If I try to write to do _CBCharacteristicWriteType.WithResponse_, I receive an error. Because I believe I shouldn't be able to read the response.
On the other characteristic, I subscribe to the notifications, but I don't know if it's working because it should only send something after I write something

Kacper20, 2016-08-20:
Hi :) 
Deciphering `CBCharacteristicProperties(rawValue: 18)`: We have `CBCharacteristicProperties` defined as follows:
```
typedef NS_OPTIONS(NSInteger, CBCharacteristicProperties) {
	CBCharacteristicPropertyBroadcast												= 0x01,
	CBCharacteristicPropertyRead													= 0x02,
	CBCharacteristicPropertyWriteWithoutResponse									= 0x04,
	CBCharacteristicPropertyWrite													= 0x08,
	CBCharacteristicPropertyNotify													= 0x10,
	CBCharacteristicPropertyIndicate												= 0x20,
	CBCharacteristicPropertyAuthenticatedSignedWrites								= 0x40,
	CBCharacteristicPropertyExtendedProperties										= 0x80,
	CBCharacteristicPropertyNotifyEncryptionRequired NS_ENUM_AVAILABLE(10_9, 6_0)		= 0x100,
	CBCharacteristicPropertyIndicateEncryptionRequired NS_ENUM_AVAILABLE(10_9, 6_0)	= 0x200
};
```
18 is `0x12` in hex. To receive this value by combining values defined above we need to connect `0x10` and `0x02`. So these three characteristics have properties `Read` and `Notify`.

Could you try read value from them?

marcelogfsouza, 2016-08-23:
Sorry for the long wait to respond here. The issue is that we didn't develop on the bluetooth device. We received like this and they told us which characteristic we should use to write and read. So I don't know exactly if it's possible to try to read from a different one, or how to make the device send data without asking it. I am pretty new with the bluetooth development, I started this month. So there is a lot of things that I don't know how to work.
The device that we received is a nRF52, and we received an extra one with the dev kit toolbox. We will try to develop something really basic just to send data to see if it's working.

Thanks

marcelogfsouza, 2016-08-25:
Hi Kacper, Thanks a lot for your help. We coded and tested with a different peripheral. So we found out that the issue was related to the other peripheral. Thanks again

closed 2016-08-25 by marcelogfsouza

Kacper20, 2016-08-25:
I'm glad to hear that :-) 
Thanks for feedback. In case of any further questions, feel free to ask.

Have a nice day :)