Polidea/RxBluetoothKit #45: CBCentralManagerState is deprecated in iOS 10

vendruscolo, 2016-08-18:


Kacper20, 2016-08-18:
Hello there,
We're aware of changes in `CoreBluetooth`. 
We'll be working on support of Swift 2.3 and 3.0 starting from the next week. Further releases of library will support iOS 10, and consist any necessary changes. 

We'll let you know about the progress in this issue. :)

vendruscolo, 2016-08-18:
Fantastic news, thanks!

For the time being, I fixed the issues by always using the `CBManagerState` type and added this type alias to \`RxCBCentralManager.swift\` file (but any file would be okay)

```
#if !swift(>=2.3)
public typealias CBManagerState = CBCentralManagerState
#endif
```

This ensures backwards compatibility with iOS 9.

The last bit is that when I compile with Xcode 8 I have to set the deployment target to iOS 10 (otherwise I'd get a ton of availability warnings). When I compile with Xcode 7 I reset the target to iOS 9.

Kacper20, 2016-08-18:
Good job :) I've looked at changes in 'CoreBluetooth' couple of weeks ago and I think it's the purest way of making it work quickly.
Are you using Swift 2.3 right now? 

Thanks for the feedback, it's really valuable.

vendruscolo, 2016-08-18:
Not really, but I have to use Xcode 8 as I’ve already upgraded to iOS 10 on my main device. But I also have to ship a version of the app that works with iOS 9.

The `#if swift()` check is there to understand if we’re targeting iOS 9, it’s not ideal and absolutely not the right way to do it, but it works.
I honestly don’t know how to handle this type rename without creating two separate classes (like the quick fix would do, by wrapping the whole class in an `@available` block).
I got that bit from here, maybe it can help us http://radex.io/xcode7-xcode8/

Kacper20, 2016-09-15:
Hi,

We've updated library to 2.0.0 version which introduces our own `BluetoothState`. It was the easiest possible way to support iOS 8+, Swift 2.2 and Swift 2.3. Contains minor API changes, like deleting deprecated and not working methods. 
The trick with typealias was not possible with keeping iOS 8.0 deployment target, unfortunately.

Hope you'll find it useful :-)

closed 2016-09-15 by Kacper20

jams032, 2018-04-13:
Does this affect on iPhone 8 , Bluetooth version 5 ? My heart rate monitor apps works fine in all iOS version but in iPhone 8 or higher i am not getting any reading.