Polidea/RxBluetoothKit #46: Notifications for value updates don't work until a subscription is made

vendruscolo, 2016-08-19:
What I'm trying to accomplish:
For my use case I have to perform multiple writes to same characteristic (e.g. sending a long string). The application splits the long message in multiple frames, then sends such frame by writing to a characteristic. The BT device reads the message and acknowledges this by updating the value of the same characteristic. At this point, my app receives the updates and proceeds to writing the next packet.

What I did:
After connecting to the peripheral I keep track of the `Characteristic` I need to use for writes:

```swift
    self.connectedPeripheral = peripheral.connect()
    self.dataCharacteristic = connectedPeripheral
            .flatMap {
                $0.characteristicWithIdentifier(MirrorCharacteristic.Data)
            }
            .flatMap {
                $0.setNotificationAndMonitorUpdates()
            }
            .shareReplay(1)
```

So far so good. When I need to send frames I have this method:

```swift
func sendData(data: NSData) -> Observable<Bool> {
    let packets = DataCharacteristicPacket.framesForData(data)

        // Turn each packet in a series of write + ack: writes a packet and then
        // waits for an ack
        let writes = packets.map { packet in
            return dataCharacteristic
                .flatMap { (characteristic: Characteristic) -> Observable<Bool> in
                    print("write")
                    return characteristic.writeValue(packet.fullPacket, type: .WithResponse)
                        .flatMap { (characteristic: Characteristic) -> Observable<Characteristic> in
                            // switch to the notification-enabled stream
                            return self.dataCharacteristic
                        }
                        .map { (characteristic: Characteristic) -> Bool in
                            // check the data we got back was an actual ack
                            // (stripped because not much of interest)
                            return true
                        }
                        .take(1)
                        .flatMap(Observable.just)
                }
        }

        // Concatenate all writes
        return writes
            .concat()
            .reduce(true) { _ in
                true // assume all writes were ok
            }
}
```

But it doesn't work. If in the init I explicitly `subscribe` to the `dataCharacteristic` it works as expected:

```swift
    self.dataCharacteristic
            .subscribeNext {
                print("value \($0.value)")
            }
```

Questions: how can I accomplish this without having to subscribe to the `Characteristic`? I think this should work just by subscribing to the return value of `sendData()` as it internally _forwards the subscription up to the original `Characteristic`_.

Any idea? What am I doing wrong?

vendruscolo, 2016-08-19:
I have another question, I don't know if it's related to this issue or not..

I have to use the `Peripheral` instance to perform writes, otherwise it doesn't work. So the code really is:

```swift
    return peripheral.writeValue(
                        packet.fullPacket,
                        forCharacteristicWithIdentifier: MirrorCharacteristic.Data,
                        type: .WithResponse
                    )
```

instead of

```swift
    return characteristic.writeValue(packet.fullPacket, type: .WithResponse)
```

(I still have to keep the original subscription to the `dataCharacteristic` around in order to receive acks).

Kacper20, 2016-08-19:
Hello,
We've looked through your implementation and there are couple of problems with it.
Basically, from what i see this code:
``` return dataCharacteristic
                .flatMap { (characteristic: Characteristic) -> Observable<Bool> in
                    print("write")```
Will never be executed, because `dataCharacteristic` will never emit any `.Next`. Notifications are working, but you're not doing anything else with `Characteristic`, so `.Next` will not be emitted.

One simple fix you could do is to change line:
`.flatMap { $0.setNotificationAndMonitorUpdates() }` 
to `.flatMap { $0.setNotificationAndMonitorUpdates().startWith($0) }

And this should do the work.
Also, my suggestion is to use for example `zip` for this purpose and zip notification & write observables, and then follow with the next one. I will try to post necessary code snippet here.

I'm not exactly sure why it's needed, because `write` always emits one item so take is not necessary, and `flatmap` does nothing aswell. 
```                        .take(1)
                        .flatMap(Observable.just)```

Also, there's no need to nest these `flatMap` operators, you could just chain them.
As i said - later today I will upload version with use of `zip` operator. For now - let me know if the fix I recommended does the work for now.

Kacper20, 2016-08-19:
To the second issue: Every call to the `writeValue` function on `characteristic` just calls function `writeValue` on `Peripheral` instance that is hold by `Characteristic`, so it's the same thing.
Could you tell a bit more about this issue?

Kacper20, 2016-08-25:
Hi there,
Did you manage it to work? How's the progress?

vendruscolo, 2016-08-26:
Hey, first of all thank you for your time, and sorry for the delay! 😅

You say that `dataCharacteristic` won't ever emit a `.Next` (because no one is subscribed to it). After the write is performed I _switch_ to it using this code:

```
peripheral.writeValue(
                        packet.fullPacketData,
                        forCharacteristicWithIdentifier: MirrorCharacteristic.AppData,
                        type: .WithResponse
                    )

                        // When writing is reported as successful, "switch" to
                        // notification stream
                        .flatMap { (characteristic: Characteristic) -> Observable<Characteristic> in
                            return self.dataCharacteristic
                        }
```

and then I `.map` the sequence to check if it received the ack. So, as far as I understand, someone should be subscribed to the stream. Isn't it?

Maybe _switching_ to the "read" stream is dead wrong and I need to try a different approach. The code for the whole process is available here https://gist.github.com/vendruscolo/0dc31f282fd34beddd4cfc3ad1542b3c

Maybe I should just create a sequence of writes, concatenating them, zipped with the acks?
Something like

```
Observable.zip(writes.concat(), dataCharacteristic) { $1 }
  .map { ... }
```

Could this be okay? Will the sequence ever complete? That's why I used the `.take(1)` bit.

Kacper20, 2016-08-26:
Hi,
I think that you should try different approach, preferably with zipping. This switching is really confusing and i think for the clarity we should stick to different approach. Now i don't have much time to actually provide you with the solution and will try to do this tomorrow. 

> You say that dataCharacteristic won't ever emit a .Next (because no one is subscribed to it). After the write is performed I switch to it using this code:

Not actually. That's not the problem with subscribing. Let's read it line by line:
```
let writes = packets.map { packet in // 1
            return dataCharacteristic // 2 
                .flatMap { (characteristic: Characteristic) -> Observable<Bool> in // 3
```

So, every packet observable you're flatmapping to the dataCharacteristic observable. This observable will fire when there's a change in the characteristic value. After that you're writing to the characteristic. I think that `dataCharacteristic` will never emit `.Next`, because this change never happens. It's because your write depends on this emitting this by `dataCharacteristic`.Next`, and nothing else changes the characteristic value, so it's just dead code from what i see.

So it's like: write depends on notification, and notification depends on write.

Have you tried this simple workaround with `startWith`?
Will provide you with the more approachable solution over the weekend or on Monday at the latest

Have a nice day.  :-)

vendruscolo, 2016-08-29:
I've tried a different approach and it seems to be working well. Nevertheless it's way more simpler. I followed your advice and I'm now zipping writes & acks together.

```swift
        let packets = AMPacket.packetsFromAppData(data)

        // Turn each packet in its own write
        let writes = appDataCharacteristic
            // AppDataCharacteristic has notifications enabled, therefore it'll
            // emit also when we receive data. We're interested in just the
            // first item to start the transfer.
            .take(1)
            .flatMap { characteristic in
                packets.map { packet in
                    characteristic.writeValue(
                        packet.fullPacketData,
                        type: .WithResponse
                    )
                }
                .concat()
            }
            .take(packets.count)

        // Listen for the acks
        let acks = appDataCharacteristic
            .map { (characteristic: Characteristic) -> Bool in
                var value: UInt8 = 0
                characteristic.value?.getBytes(&value, length: 1)
                return value == AMPacket.Flag.Ack.rawValue
            }
            .take(packets.count)

        // Perform all writes, and interleave them with an ack.
        // After that count the number of acks (we want all of them to be
        // successful in order to call it a day)
        return Observable.zip(writes, acks) {
            $1
        }
            .reduce(0) { accumulated, item in
                accumulated + (item == true ? 1 : 0)
            }
            .map { count in
                if count != packets.count {
                    throw NearbyMirrorError.DataTransferFailed
                }
            }
```

I'm not sure about those `.take(packets.count)` but without those my sequence won't complete. Any suggestion?

Beside that, I'm really happy about the outcome. It's quite easy to understand and most importantly it uses the same technique (zipping) I use for the _opposite_ protocol (I had to implement the functionality of my app receiving arbitrary data and writing back acks).

Kacper20, 2016-08-29:
Hi,
Here: http://stackoverflow.com/questions/39206622/how-to-combine-write-characteristics-and-charcteristics-notify-using-rxbluetooth Here you have @Cierpliwy answer to very similiar question.
Basically it's doing sequentional writes that are waiting on the notification.
It should work and is probably easier to understand and maintain.

Let us know how it works for you :-)

vendruscolo, 2016-09-13:
Thank you. Splitting the process to have a sequence mad of _write+ack_ pairs definitely clears things up, but I'm afraid that race conditions could happen..?

At the moment the last approach I came up with works fine and I don't want to touch it again :P in the future I might end up following the example you provided, mainly because it seems easier to understand.

Thank you again for this library and for the time you took to help me! 👍

Kacper20, 2016-09-13:
If you're doing work on only one thread this should be just fine.

No problem, ask again in case of any issues :-)

closed 2016-09-13 by Kacper20