Polidea/RxBluetoothKit #53: Reading Multiple Characteristics

z4gunn, 2016-09-07:
I am having difficulty reading multiple characteristics by combining observables.  For some reason the Completed event fires before the values are read from the device.

`    
  
        peripheral.peripheral.connect()
            .flatMap
            {
                Observable.combineLatest(
                    $0.readValueForCharacteristicWithIdentifier(BLEDeviceCharacteristic.ManufacturerName),
                    $0.readValueForCharacteristicWithIdentifier(BLEDeviceCharacteristic.ModelNumber),
                    $0.readValueForCharacteristicWithIdentifier(BLEDeviceCharacteristic.SerialNumber))
                {
                    (mfg, model, serial) in
                    
                    print("Manufacturer = \(mfg)")
                    print("Model Number = \(model)")
                    print("Serial Number = \(serial)")
                }
            }
            .subscribe { event in
            
                switch(event) {
                    
                case .Next(let value):
                    print("Next Value = \(value)")
                    
                case .Error(let error):
                    print("Error = \(error)")
                    
                case .Completed:
                    print("Completed")
                    
                }
            }
            .addDisposableTo(disposeBag)
`

z4gunn, 2016-09-07:
I also tried using the "zip" operator with the same results.

z4gunn, 2016-09-07:
Reopening issue

Kacper20, 2016-09-13:
Hi there,
Sorry for the delay in answering. It shouldn't be like that. To help us investigate the problem could you  make a breakpoints on lines that have this line of code: `return Observable.empty()` in `Peripheral.swift` and let us know if any of this lines is called?

Thanks for help

Kacper20, 2016-09-13:
I'm talking about two lines: one in `discoverIncludedServices` and another in `discoverCharacteristics`.

z4gunn, 2016-09-13:
I could not find the Observable.empty() calls in the functions that you mentioned, but realized I was using v1.1.1.  I just updated to v1.2.3 and it now seems to working fine (i.e. multiple characteristics are being returned before the complete). 

Thanks!

Kacper20, 2016-09-13:
We've fixed couple of bugs since 1.1.1. Glad it works for you now. 👍

closed 2016-09-13 by Kacper20