Polidea/RxBluetoothKit #55: Library won't compile with Xcode 8

mane87, 2016-09-14:
Hi,
i 've tried the main branch and the feature/swift-2.3 with the "Use Legacy Swift Language Version" set to "Yes".
I have also tried feature/swift-3.0 with this flag set to "No". 

I was not able to get any of the above settings to work. Am i doing something wrong?

Thanks, m

Kacper20, 2016-09-14:
Hi,
We're currently working on support for Swift 2.3 and it'll be released today, preferably in a couple of hours. Will let you know when it happens - after that you should be able to adapt Xcode 8 & Swift 2.3 support. 
Swift 3.0 support will be released a bit later(we'll be working on it starting from today), as it requires more work and thinking from our side.

Kacper20, 2016-09-14:
@mane87 I have couple of questions to you:
How are you integrating dependencies? Is it Cocoapods or Carthage?
Is this project created with Xcode 8 or earlier?
Have you added SWIFT_VERSION to user defined settings in your xcodeproj?

Kacper20, 2016-09-14:
I've just tried to integrate sample proj I've created with Xcode 7.3, later integrated it with pods and xcode 8. Everything works just fine.
If you're using Cocoapods try to use this little script in your `Podfile` for 2.3 support.:
```
 post_install do |installer|
    installer.pods_project.targets.each do |target|
        target.build_configurations.each do |config|
            config.build_settings['SWIFT_VERSION'] = '2.3'
        end
    end
  end
```
Also, are you downloading version from branch like that?:
 ```
  pod 'RxBluetoothKit', :git => 'https://github.com/Polidea/RxBluetoothKit.git', :branch => 'feature/swift-2.3'
````

Let us know how it works for you. Version from branch is development one but should work as of now :)

mane87, 2016-09-14:
@Kacper20 

I am using cocoapods and created the project with xcode 7.

I'll give it a try in the next few days (traveling right now)

Thank you again for the super-fast response! You guys are amazing!!!👍🏼👍🏼👍🏼

Kacper20, 2016-09-15:
Hi there.
We've released version 2.0.0 of the library. It supports Swift 2.3, contains minor API changes which shouldn't be a problem to adapt. I've tried it a while ago and it works well with project created on Xcode 7 and opened in Xcode 8. Only things to worry about is to be sure to have `Use Swift Legacy version` set to `YES` and paste this script I gave you earlier in your `Podfile`.

Thanks for feedback and let us know how it works for you.

mane87, 2016-09-16:
Hi,
it worked perfectly!

Thanks for the great work!

closed 2016-09-16 by mane87