Polidea/RxBluetoothKit #63: Compiler errors while installing latest version for Swift 3 (Xcode 8)

GauravTechbirds, 2016-10-04:
After installing it in swift 3 using  pod 'RxBluetoothKit', Compiler shows me 53 errors in Xcode 8.

Anyway how can I install this successfully in Xcode 8?

Kacper20, 2016-10-04:
Hi there. Which version of RxSwift is downloading while you're running pod install? 
I recommend you to add this piece to your Podfile:
``` post_install do |installer|
    installer.pods_project.targets.each do |target|
        target.build_configurations.each do |config|
            config.build_settings['SWIFT_VERSION'] = '3.0'
        end
    end
  end```
and run `pod update`. 

Let me know how it worked out for you :-)

GauravTechbirds, 2016-10-04:
Thanks @Kacper20 

I had to do replace  **pod 'RxBluetoothKit'** with **pod 'RxBluetoothKit', :git => 'https://github.com/Polidea/RxBluetoothKit.git', :branch => 'feature/swift-3.0'**

Kacper20, 2016-10-04:
@GauravTechbirds This is not perfect solution, because this branch is already merged into master, and we'll delete it soon. So if you want to be sure to have always-working solution i encourage you to get version from master. 
`pod update` should do the work and download 3.0.1 version of the library for you. This version supports Swift 3.0. 
In case if something would not work - please attach log received by running `pod update` command :)

closed 2016-10-05 by Kacper20