Polidea/RxBluetoothKit #66: Support for CBPeripheralDelegate

Kacper20, 2016-10-06:
To cover entirely CoreBluetooth API we need to support part of it that makes possible to use Apple device as a peripheral. 
I'm not entirely sure how it'll affect the other parts of API - since in iOS 10 `CBPeripheralManager` and `CBCentralManager` are inheriting from `CBManager` subclass it can be a good idea to make such move in our library too. This would probably mean renaming `BluetoothManager`, and introduce breaking change. 

Need to perform further investigations of possibilities to make it as smooth as possible.

mkulikov, 2016-11-19:
it will be great!

garsdle, 2017-02-22:
How is this coming along? I'm actually using both CBCentralManager and CBPeripheralManager in a pet project and was going to make the jump to RxBluetoothKit. Having both would be amazing. I'm willing to help as well if there is anything I can do.

Kacper20, 2017-10-22:
I am closing this issue — will create a new one soon with detailed design.

closed 2017-10-22 by Kacper20