Polidea/RxBluetoothKit #70: Observable.from error in Swift 3 app

jwhitley, 2016-10-26:
I've have an app I'm converting from Swift 2 to Swift 3, using code that's pretty much identical to the discoverCharacteristics idiom from the README:

```
Observable.from($0.discoverCharacteristics([my_characteristic_uuid]))
```

Xcode 8.0 is producing the following error against the parameter of `Observable.from()` in the above:

```
/path/to/our/project/XferControlMessage.swift:109:43: 
Cannot convert value of type 'Observable<[Characteristic]>'
 (aka 'Observable<Array<Characteristic>>') to expected argument type '[_]'
```

This error reads like the compiler perhaps isn't choosing the `from<S: Sequence>` variant defined in `Observable+Creation.swift`?  As I understand, the whole point of `Observable.from` is to convert `Observable<[Characteristic]>` into an observable sequence `Observable<Characteristic>`.  So this definitely seems like either a code or documentation bug, but I'm new enough to Swift, and especially the nuances of Swift 3, that I'm not immediately clear on the root cause here.

This is using RxBluetoothKit 3.0.4 and RxSwift 3.0.0.

Kacper20, 2016-10-26:
Hi,

We've dropped support for our own `Observable.from` operator, because in 3.0 version RxSwift contains similiar one that is called the same way.

Information in our README.md is not up to date, we've missed it and I will fix it right away.
Now, to fix it if you have specified services/characteristics I would advice you to follow `Convenience calling methods` section of our README.

To use `Observable.from` from RxSwift you need to pass to it array of elements, instead of observable on array.
So it would look like:
```peripheral.connect()
    .flatMap { $0.discoverServices([serviceId]).flatMap { Observable.from($0) } }```
Or writing it one level lower:
```
peripheral.connect
.flatMap { $0.discoverServices([serviceId]) }
.flatMap { Observable.from($0) }
```

This should do the work.
To make it nicer you could write your own extension on Observable, which basically calls `Observable.from($0)` but maybe it would be nicer to have it in more meaningful way.

I'm jumping to README and updating it now. :) 

Thanks again and let me know how it works for you :)

Kacper20, 2016-10-26:
It's fixed and in master right now :)

jwhitley, 2016-10-26:
Fantastic, that resolved it. Thanks so much for the prompt reply!  I'll leave off adding an extension until I've gotten more experience with RxSwift & RxBluetoothKit.

closed 2016-10-26 by jwhitley