Polidea/RxBluetoothKit #71: warnning:

swieer, 2016-10-27:
[CoreBluetooth] API MISUSE: <private> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported

Kacper20, 2016-10-27:
Could you provide a snippet of code to clarify things?

swieer, 2016-10-27:
@Kacper20 ,use Swift3.0, iOS10.0.2, Example project. running on iPhone5s.
Sorry, My English is poor
Thanks!

Kacper20, 2016-10-27:
Hi there,
We think that there's nothing to be worried about. We have delegate which implements all of the `CBCentralManagerDelegate` methods. Since you're not using state restoration, and we implement all of the methods - this warning occurs. 
We will think about a way to get rid of this warning, but this is nothing you should be worried about.

Does this answers your concerns?
Thanks for feedback :)

closed 2016-10-28 by Kacper20

z4gunn, 2017-04-22:
If you plan in using BLE state restoration then you have to check "Uses Bluetooth LE accessory" in the "Background Modes" section of the Capabilities tab, which will silence this error.  Also this warning becomes an error in Swift 3.2 / Xcode 8.3.

[image]