Polidea/RxBluetoothKit #75: Notification issue merging to Swift 3.

timbjarengren, 2016-11-11:
Hi, I'm currently updating my Swift code to the Swift 3 and I'm now experience different behaviour even though the code structure haven't changed. I am using RxBluetoothKit to write settings to a beacon and then receiving the returning notification.

The following is the code which worked in RxBluetoothKit 2.6 and Swift 2.3, but not now with RxBluetoothKit 3.0.1 and Swift 3:

    ...

    var disposeBag: DisposeBag?
    var peripheral: Peripheral?
    var characteristic: Characteristic?

    func start(peripheral: Peripheral, blocks: [[UInt8]]) {
        
        disposeBag = DisposeBag()
        self.peripheral = peripheral
        
        let timeout = Date(timeIntervalSinceNow: timeoutInSeconds)
        
        peripheral
            .connect()
            .flatMap { peripheral in
                peripheral.discoverServices([self.serviceUuid])
            }
            .flatMap { Observable.from($0) }
            .flatMap { service in
                service.discoverCharacteristics([self.characteristicUuid])
            }
            .flatMap { Observable.from($0) }
            .takeWhile { _ in NSDate().laterDate(timeout) == timeout }
            .take(1)
            .do(onNext: { characteristic in
                debugPrint("Found characteristic")
                self.characteristic = characteristic
                self.setWriteNotification()
                Observable.from(blocks).subscribe(onNext: { self.writeBlock(block: $0) }).addDisposableTo(self.disposeBag!)
            })
            .subscribe()
            .addDisposableTo(disposeBag!)
    }

    fileprivate func setWriteNotification() {
        
        characteristic?
            .setNotificationAndMonitorUpdates()
            .subscribe(onNext: { (characteristic) in
                debugPrint("Did receive characteristic update")
                if let value = characteristic.value {
                    // Do stuff
                }
            }, onError: { (errorType) in
                debugPrint("Did disconnect")
            }).addDisposableTo(self.disposeBag!)
        
    }
    
    fileprivate func writeBlock(block: [UInt8]) {
        debugPrint("Write next block")
        characteristic?
            .writeValue(NSData(bytes: block, length: block.count) as Data, type: .withResponse)
            .subscribe()
            .addDisposableTo(disposeBag!)
    }

    func close() {
        disposeBag = nil
        
        _ = characteristic?.setNotifyValue(false).subscribe()
        characteristic = nil
        
        _ = peripheral?.cancelConnection().subscribe()
        peripheral = nil
    }

    ...

The output to the console is:

    Found characteristic
    Write next block
    Did disconnect

The `"Did disconnect"` is received after 10 seconds, since the beacon automatically closes the connection after no activity in that time. However, I should not get that message since I expect a notification so I then can call `close()`.

Do I do something wrong here? This process worked before and the only thing I have done is updating it to the current syntax.

Thanks!

Kacper20, 2016-11-11:
Hi,

We will study your code and give you answer on Monday.
In the library itself nothing has changed in logic between Swift 2.3 and Swift 3 version, at least nothing I am aware of. If it's a bug we'll make sure to fix it as soon as possible. 

Thanks for feedback and reporting this issue,

Kacper

Kacper20, 2016-11-14:
Hi,

We've read through the code and would like to propose some improvements:

First, in order to make it simple and more clear, we encourage you to read and use the code from **Convenience calling methods** section in README.md - it'll simplify things a bit.
Another thing is to replace this piece of code:
`.takeWhile { _ in NSDate().laterDate(timeout) == timeout }`
with built-in RxSwift `timeout` operator - this will do the right job.

To understand more about your error we need to know what the exact error is, so please add to your call: 
`debugPrint("Did disconnect")` printing the error type you've received. This way we'll be able to help you debug your issue.

Kacper

timbjarengren, 2016-11-14:
Hi,

Thanks for your improvement tips. I have changed to using the "Convenience calling methods" instead and changed to the timeout on RxSwift, and yes, the code become more clear.

When I changed to use the convenience method, I got my notification callback as desired. However, I then instead got another issue. Since I write an arbitrary number of blocks to the beacon, I need to make sure that I write them in the correct order. But that is not the case. If I have five blocks to write, the app sends them in the order: 5,1,2,3,4. I have verified that I write them to the peripheral in the correct order (1,2,3,4,5), but when the debugger get to row 289 in `Peripheral.swift`, they are in the wrong order (5,1,2,3,4).

The new code is as follows:

    func start(peripheral: Peripheral, blocks: [[UInt8]]) {
        
        disposeBag = DisposeBag()
        self.peripheral = peripheral
        
        peripheral
            .connect()
            .timeout(timeoutInSeconds, scheduler: MainScheduler.instance)
            .flatMap { (peripheral) -> Observable<[UInt8]> in
                self.setWriteNotification()
                return Observable.from(blocks).asObservable()
            }
            .subscribe(
                onNext: { (block) in
                    self.write(block: block)
                }
            )
            .addDisposableTo(disposeBag!)
    }
    
    fileprivate func setWriteNotification() {
        
        peripheral?
            .setNotificationAndMonitorUpdates(for: MyCharacteristic.authorization)
            .subscribe(onNext: { (characteristic) in
                debugPrint("Did receive characteristic update")
                if let value = characteristic.value {
                    // Do stuff
                }
            }, onError: { (errorType) in
                debugPrint("Did disconnect")
            }).addDisposableTo(self.disposeBag!)
        
    }
    
    fileprivate func write(block: [UInt8]) {
        debugPrint("Write next block")
        
        let data = NSData(bytes: block, length: block.count) as Data
        
        peripheral?
            .writeValue(data, for: MyCharacteristic.authorization, type: .withResponse)
            .subscribe(
                onNext: { characteristic in
                    debugPrint("Did wrote next block")
                }
            ).addDisposableTo(disposeBag!)

    }
    
    func close() {
        disposeBag = nil
        _ = peripheral?.cancelConnection().subscribe()
        peripheral = nil
    }

This gives me the output:

    Write next block
    Write next block
    Write next block
    Write next block
    Write next block
    Did wrote next block
    Did wrote next block
    Did wrote next block
    Did wrote next block
    Did wrote next block

However, when validating the value in the `// Do stuff`block, I can see that the values have been written in a different order than desired.

Is this expected behaviour and do I need to write the next block when I get the `Did wrote next block` callback? Or, do you have any ideas or pointers to how I could attack this issue? I need to write a arbitrary number of blocks in a given order.

About the `Did disconnect` error callback on the previous procedure. The error received was a disconnected-type and that was expected since the beacon disconnect itself after about 10 seconds. 

Tim B

Cierpliwy, 2016-11-14:
Hello,

Let me add more suggestions to your code and explain what really happens. Bluetooth operations are asynchronous and don't block even when executing them on the same thread. All observable subscriptions are executed immediately. That's why, by using operator `flatMap` which doesn't guarantee synchronous execution, you have mixed results.

I would start from making your write method to return Observable:
```
fileprivate func write(peripheral: Peripheral, block: [UInt8]) -> Observable<Characteristic> {
    let data = NSData(bytes: block, length: block.count) as Data
    return peripheral.writeValue(data, for: MyCharacteristic.authorization, type: .withResponse)
}
```

It so small that I would consider inlining it. Then let's update your code to run synchronously:

```
 peripheral
        .connect()
        .timeout(timeoutInSeconds, scheduler: MainScheduler.instance)
        .flatMap { (peripheral) -> Observable<[UInt8]> in
            self.setWriteNotification() // Consider merging it to the flow
            return Observable.from(blocks).asObservable() // Do you really need asObservable?
        }
        .map { (block) -> Observable<Observable<Characteristic>> in // Maybe you can omit types here
             return write(peripheral, block); // Inline?
        }
        .concat()
        .subscribe()
        .addDisposableTo(disposeBag!)
```

I still don't like calling setWriteNotification() in flatmap and personally will try to merge it somehow in flow. We changed flatMap to map + concat. Map's job is to return stream of observables (or your operations) in proper order. Then we need to concat it to emit elements from inner observables: https://github.com/ReactiveX/RxSwift/blob/3c55a309a24fbe3dbc480431798eec072b633b85/RxSwift/Observables/Observable%2BMultiple.swift#L129

I didn't test this code and write it without IDE, but I think it should work.

timbjarengren, 2016-11-15:
Hi,

This is really interesting! I don't think I have really understand the power of ReactiveX and how to use it fully and correctly. I have updated according to your ideas and it worked! I only needed to change the return type of the map to just `Observable<Characteristic>`. The `asObservble()` could be removed. It was a rest from when I tried to fix some error. The problem was however that the compiler couldn't figure out the correct return type, so I needed to add that. I also made the writing inline. Much cleaner!

I will now dig deeper and see if I can try to include the notification in the flow.

Thank you guys for your help. I really appreciate that!

closed 2016-11-15 by timbjarengren

Kacper20, 2016-11-15:
This post: http://stackoverflow.com/questions/39206622/how-to-combine-write-characteristics-and-charcteristics-notify-using-rxbluetooth by @Cierpliwy could be also useful to you when doing sequentional job.

timbjarengren, 2016-11-15:
Sound exactly like what I want to do. Thanks!

LiuSky, 2016-11-16:
Excuse me @timbjarengren ,Can you the ultimate solution to stick out.

timbjarengren, 2016-11-17:
@LiuSky I havn't got time to merge together the notification with the writing yet, so I don't have my final solution yet. So currently it is the one above.

LiuSky, 2016-11-18:
@timbjarengren  thank you