Polidea/RxBluetoothKit #76: Because bluetooth transmission data of each time is limited, if separated, can have what good example?

LiuSky, 2016-11-16:


Kacper20, 2016-11-16:
Hi there,
Basically you need to split your data into the packets.
The best way would be to have a function which splits it and returns an Array of NSData values.
Then, you need to perform sequentional write. Code depends on the way your characteristic is working on(do you need to wait on the notification, or just perform writes with acknowledge?).
Here you have example of very similiar issue(seq writes) and possible solution:
https://github.com/Polidea/RxBluetoothKit/issues/75

If case you need more help - please feel free to ask.

Kacper

LiuSky, 2016-11-16:
All right.I go to take a look at first.thank you @Kacper20

LiuSky, 2016-11-16:
[image]
I found that the print soon, or accidentally messing with the order.This is what circumstance @Kacper20

closed 2016-11-18 by LiuSky