Polidea/RxBluetoothKit #79: Change internal structure of BluetoothManager/Peripheral/Service

Kacper20, 2016-11-24:
There's proposal by @Cierpliwy to change internal structure of BluetoothManager/Peripheral/Service instances. This will allow us to remove complexity which is now caused by nested objects that are called similiar, and is probably unnecessary.

Cierpliwy, 2016-11-24:
This is a new proposition which shows how we can structure our code in a new way to allow better testability and possibility to use implementation specific things if possible.

Current architecture consists of three major components
- Implementation protocols in form of: PeripheralType, CharacteristicType
- Generic implementation classes such as: Peripheral, Characetristic and so on which
  operates on above interfaces and includes them by composition
- Implementation specific classes which implement protocols.

This architecture has few issues:
- We need type casting when implementing protocols (for example):
```swift
func discoverIncludedServices(_ includedServiceUUIDs: [CBUUID]?, for service: RxServiceType) {
     peripheral.discoverIncludedServices(includedServiceUUIDs, for: (service as! RxCBService).service)
}
```
- Generic classes were still influenced by implementation classes in minor way. For example
  a need to add dispatch queue parameter in BluetoothManager constructor.
- We couldn't access implementation specific filed even when we knew it was allowed without
  ugly casting.
- We could mix different implementations which will lead in most cases to crashes.

New proposition solves above problems:
- Simpler architecture which contains only Protocols and Implementation classes.
- Generic code is added as extensions to Protocols
- By using concrete instance you still have access to concrete types which depend on it.
- You cannot mix different implementations.

Simple snippet which presents architecture:
```swift
// Basic types

protocol CentralManagerType {
    associatedtype Peripheral: PeripheralType
    var peripherals: [Peripheral] { get }
}

protocol PeripheralType {
    associatedtype Service: ServiceType
    func connect() -> Self
    var services: [Service] { get }
    var id: Int { get }
}

protocol ServiceType {
    var id: Int { get }
}

// Common implementation

extension CentralManagerType {
    func connectTwiceToAllPeripherals() {
        for peripheral in peripherals {
            peripheral.connectTwice()
        }
    }
}

extension PeripheralType {
    func connectTwice() -> Self {
        connect()
        connect()
        return self;
    }
}

// Test specific implementation

class TestCentralManager: CentralManagerType {
    internal var peripherals: [TestPeripheral] = [TestPeripheral()]
    typealias Peripheral = TestPeripheral
}

class TestPeripheral : PeripheralType {
    internal var id: Int = 9
    internal var services: [TestService] = [TestService()]

    internal func connect() -> Self {
        print("Test connect")
        return self;
    }
    typealias Service = TestService
}

class TestService : ServiceType {
    internal var id: Int = 10
}

// CB Sepcific implementation

class CBCentralManager: CentralManagerType {
    internal var peripherals: [CBPeripheral] = [CBPeripheral(), CBPeripheral()]
    typealias Peripheral = CBPeripheral
}

class CBPeripheral: PeripheralType {
    internal var id: Int = 0
    typealias Service = CBService
    internal var services: [CBService] = [CBService()]

    func implFunc() {
        print("Impl func")
    }

    internal func connect() -> Self {
        print("CBCOnnect!")
        return self;
    }
}

class CBService: ServiceType {
    var id: Int { return 3}
}

// Using specific classes allows to use impl detail if needed (for example access to inner CBPeripheral)

func testSpecific(cbCentralManager: CBCentralManager) {
    print("SPECIFIC TEST: --------------------")
    cbCentralManager.connectTwiceToAllPeripherals();
    cbCentralManager.peripherals.forEach { peripheral in
        peripheral.implFunc()
    }
}

testSpecific(cbCentralManager: CBCentralManager())

// Using generic tests is harder...

func testGeneric<P: PeripheralType, S: ServiceType, C:CentralManagerType>(centralManager: C) where C.Peripheral == P, P.Service == S  {
    print("GENERIC TEST: --------------------")
    centralManager.connectTwiceToAllPeripherals()
    centralManager.peripherals.forEach { peripheral in
        peripheral.services.forEach { service in
            print("\(service.id)")
        }
        peripheral.connect()
    }
}

// But it can be used with any impl
testGeneric(centralManager: CBCentralManager())
testGeneric(centralManager: TestCentralManager())
```

Kacper20, 2017-10-22:
We’ve decided that this is not a way to go, since approach is too complicated and requires a lot of type erasure, which could be inconvenient to the users. We have something new in mind that will be described in details soon.

closed 2017-10-22 by Kacper20