Polidea/RxBluetoothKit #81: Support for multiple service instances with same UUID

abeintopalo, 2016-11-28:
Hello,

First of all I'd like to thank you for your great work to make this awesome library.
I really like the API and it is a pleasure to work with. :)

I have a use case when the peripheral has multiple instances of a BLE service (i.e. multiple service instances with same UUID).
Each service instance have the same set of characteristics.

I have tried the following code to test if RxBluetoothKit can handle this scenario.
The code below supposed to read the value of the same characteristic for each service instances:

```
peripheral.connect()
    .flatMap { $0.discoverServices([serviceId]) }
    .flatMap { Observable.from($0) }
    .flatMap { $0.discoverCharacteristics([characteristicId])}
    .flatMap { Observable.from($0) }
    .flatMap { $0.readValue() }
    .subscribe(onNext: { characteristic in
        guard let data = characteristic.value else { return }
        let dataString = NSString(data: data, encoding: String.Encoding.utf8.rawValue)!
        print("Value: \(dataString)")
    })
```

Unfortunately, the code above prints the characteristic value only for the first service instance.
The value printing code is invoked as many times as many instances of the same service has been discovered but
it seems that the characteristic passed to "subscriber(onNext:)" will be connected only to the first discovered service instance.

I would like to ask if you know any way to get this working when there are multiple service instances with same UUID?
Any help is appreciated.

Kacper20, 2016-11-29:
Hi there,

Sorry for the late response - we were thinking about it today and unfortunately there's no easy answer to this question. Currently that behavior is not supported by the lib. What we're doing internally in library is manually closing the stream after receiving first service/characteristic that fits given UUID.

We think that it's major issue now, and we're focusing on fixing this thing. We thought about the specific design and unfortunately we're unable to support this in 3.x version - it would break compatibility for other people that are using the library. Also - to make it work right and independent of CoreBluetooth callback implementations we would need to cache `CBService` objects internally(because as we believe in this case it's the only way to distinguish between service instances). Also - we need tochange the Rx stream nature in case of services discover(`discoverService` should never complete I suppose).

I've created an issue for this: #82, and we're working to support this in newest, 4.0 version.

If you need this feature right away we could help you develop some kind of workaround until the 4.0 version will be released.
Thank you very much for feedback,

Kacper

abeintopalo, 2016-11-30:
Hi Kacper,

Thank you for response. I really appreciate that you are planning to fix this in the next release.
I've just learned that in addition to multiple instances of services with same UUID I need to deal with multiple instances
of characteristics with same UUID. :) Would it possible to add that support too?

Some background info of my use case:

Service instances would be differentiated by the _Namespace_ field of Characteristic Presentation Format descriptor similarly as in standard battery service: [Battery service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.battery_service.xml)
So, basically each characteristic would have a service specific value set in the namespace field of CPF descriptor.

Characteristics with same UUID would be differentiated by the _Description_ field of CPF descriptor.

Pretty weird case I know... :)

Do you have any estimate or plan when do you target to release the first version of 4.0?

Attila

Kacper20, 2016-11-30:
Hi @abeintopalo,

Of course. I haven't mentioned it explicitely but we have this in mind - it should work in exact the same way for characteristics. 

It's great that you posted specifics about your use case.
We will setup mock device that works in similar way and will test implementation on this.
Before the release I will probably ask you to get lib from branch and test this for us aswell on real device. :)

Regarding dates - we have the scope of 4.0 ready and basic planning behind us so now we're jumping right to the implementation. You could expect it in coming weeks. Don't want to make promises right now because of the coming christmas.
 Will surely update you with more specific date in a couple of days when we'll have some major architectural changes behind us :)

Thank you for feedback
Kacper,

abeintopalo, 2016-12-01:
Hi Kacper,

That's a very good news. I'm happy to test it when you have a testable version. :)

Br,
Attila

closed 2017-05-03 by abeintopalo