Polidea/RxBluetoothKit #83: Can I set PIN for Bluetooth Pairing Request programmatically (not manualy)?

mkulikov, 2016-11-30:
http://joxi.ru/Vm6QkZ5CDDdNer

Kacper20, 2016-11-30:
Hi there,

From what you've said I understand that your peripheral has some kind of encrypted characteristic and uses SSP to protect it. What kind of authentication mechanism SSP uses? This is crucial to decide what options do we have. CoreBluetooth by itself does not provide method to write PIN(nor any other form of auth) - it's basicallly built on different layer. 

I'll be able to help if you'll provide a bit more information about the whole process: iOS versions, bluetooth version, SSP authentication type etc.

Thanks!

Kacper20, 2016-11-30:
Also some additional info:
From what I've understood you're doing PIN based pairing. Have you looked into `JustWorks` pairing?
Does your sensor have input or output capabilities?(lack of it is required by the `JustWorks` pairing).

Kacper

dmkuzmin, 2016-12-02:
Hi, Kacper20.

We use a peripheral device with the following pairing options:
1.) Peripheral device is the initiator of the pairing;
2.) Peripheral device supports bluetooth 4.2;
3.) Use Passkey Entry pairing;
4.) IO capabilities of peripheral device - Display Only;
5.) support MITM;
6.) use Bonding;

Kacper20, 2017-04-26:
Hello @dmkuzmin 
Have you managed to resolve the issue somehow?

Regards,

Kacper

Kacper20, 2017-05-04:
Closing for now. Please feel free to reopen in case of any further questions.

Regards,
Kacper

closed 2017-05-04 by Kacper20