Polidea/RxBluetoothKit #84: Possible race condition while scanning

dariuszseweryn, 2016-12-01:
Repeated calling of `BluetoothManager.scanForPeripherals()` sometimes causes the scan to be stalled.

Kacper20, 2017-01-18:
What's the progress on this @Cierpliwy ?

cfilipov, 2017-02-17:
I'm also having an issue with `scanForPeripherals()` not working on successive calls. In my case, I am trying to retrieve a connected peripheral, then scan for five seconds, then attempt to retrieve a connected peripheral again and repeat until a peripheral is found. To achieve this I am using the `timeout` along with the `retry` operators. The very first scan begins as expected, but subsequent subscriptions to scan do not result in any advertisements. 

Stepping through [`BluetoothManager.swift`](https://github.com/Polidea/RxBluetoothKit/blob/dd0c417bcc1a7cbb0e7d3b7ae1e07286a6de5ad6/Source/BluetoothManager.swift) I was able to observe the following:

1. On the very fisrt call to `BluetoothManager.scanForPeripherals()` a scan is added to the scan queue and `centralManager.scanForPeripherals()` is called, as expected [[165]](https://github.com/Polidea/RxBluetoothKit/blob/dd0c417bcc1a7cbb0e7d3b7ae1e07286a6de5ad6/Source/BluetoothManager.swift#L165)
2. The timeout occurs, causing the observable to error
3. The `retry` operator catches the error and re-subscribes to the scan observable
4. The observable from the previous scan is found [[134]](https://github.com/Polidea/RxBluetoothKit/blob/dd0c417bcc1a7cbb0e7d3b7ae1e07286a6de5ad6/Source/BluetoothManager.swift#L134) in the scan queue and re-used [[136]](https://github.com/Polidea/RxBluetoothKit/blob/dd0c417bcc1a7cbb0e7d3b7ae1e07286a6de5ad6/Source/BluetoothManager.swift#L136). (https://github.com/Polidea/RxBluetoothKit/blob/dd0c417bcc1a7cbb0e7d3b7ae1e07286a6de5ad6/Source/BluetoothManager.swift#L135)
5. Finally, the dispose callback is called, which stops the scan and removes the single item (which has just been re-used) from the queue [[167]](https://github.com/Polidea/RxBluetoothKit/blob/dd0c417bcc1a7cbb0e7d3b7ae1e07286a6de5ad6/Source/BluetoothManager.swift#L167)
6. We are now on the second "scan" however, because this scan re-used the operation from the original scan it has already been stopped and no scan is actually taking place.

Basically, if you attempt to scan before a previous identical scan (in terms of UUIDs) is completed, the operation will be re-used and the second attempt to scan will be halted along with the first.

As a workaround I was able to introduce a delay which seems to cause the dispose to happen before the next subscribe, so subsequent scans will work.

Below is a minimal example demonstrating the issue. Uncommenting the delay line fixes the issue.

```Swift
override func viewDidLoad() {
    super.viewDidLoad()
    
    let manager = BluetoothManager(queue: .main)
    let scanOpts = [CBCentralManagerScanOptionAllowDuplicatesKey: NSNumber(value: true)];
    
    let scannedPeripherals = manager.scanForPeripherals(withServices: nil, options: scanOpts)
        .do(onSubscribe: {
                pint("Searching...")
            })
        .map { (name: $0.peripheral.name ?? "[no name]", scannedPeripheral: $0) }
        .do(onNext: {
            print("\tFound: \($0.name), RSSI: \($0.scannedPeripheral.rssi)")
        })
        .filter(isCloseEnough)
        .take(1)
        .map { $0.scannedPeripheral.peripheral }
        .timeout(5.0, scheduler: MainScheduler.instance)
    
    let _ = manager.rx_state
        .filter { $0 == .poweredOn }
        //.delay(0.01, scheduler: MainScheduler.instance) // <- Workaround
        .flatMap { _ in manager.retrieveConnectedPeripherals(withServices: [serviceId]) }
        .flatMap { $0.isEmpty ? scannedPeripherals : Observable.from($0) }
        .flatMap { $0.connect().timeout(5, scheduler: MainScheduler.instance) }
        .do(onNext: {
            print("Connected: \($0.isConnected)")
        })
        .retry()
        .subscribe()
        .disposed(by: bag)
}
```

Kacper20, 2017-02-17:
Hello,

Could you try to get library from branch `queue_subscribe_on_fix` and try whether it fixes your issue?
It helped with scenario that @dariuszseweryn had and we want to be really sure before we'll merge it into master.

Thank you very much for feedback :)

cfilipov, 2017-02-17:
Unfortunately, using the branch `queue_subsribe_on_fix` did not resolve the issue for me (using the above sample code).

Kacper20, 2017-02-19:
Thanks @cfilipov. 
We're going to investigate further on this issue and will let you know.

cfilipov, 2017-02-21:
Thanks, let me know if there's any way I can help.

askielboe, 2017-05-03:
Seeing the same issue. Inspired by @cfilipov (thanks!) I'm working around it by calling `.repeat()` using a `RepeatBehaviour.delay`.

This will do a rescan every ~ 5 seconds for ~ 10 minutes before throwing an error:

    .timeout(4, scheduler: MainScheduler.instance)
    .retry(RepeatBehavior.delayed(maxCount: 120, time: 1))

Kacper20, 2017-10-26:
We're working on fixing this in 5.0 issue by simplifying our scan logic a lot. I will keep you posted on this. Thanks for patience — I know it's been a long time.

paweljaneczek, 2018-01-24:
In #206 we have simplified scanning logic - removed all complicated logic related to scanning queueing and reusing. It should fix that problem.
Now, when user tries to scan when there is scan in progress  it will return `BluetoothError.scanInProgress` error.

closed 2018-01-24 by paweljaneczek