Polidea/RxBluetoothKit #86: writeValue accepting CharacteristicIdentifier method is missing

alfishe, 2016-12-21:
There is only Peripheral.writeValue with DescriptorIdentifier. But it's impossible to writeValue for Characteristic using convenience helpers

Kacper20, 2016-12-21:
Hi there,

Which version of the library you're using?
In file `Peripheral+Convenience.swift` we have method: 
```
    public func writeValue(_ data: Data, for identifier: CharacteristicIdentifier,
                    type: CBCharacteristicWriteType) -> Observable<Characteristic> 
```
Isn't that what you're looking for?
Thanks for feedback,

Kacper

alfishe, 2016-12-22:
Yes, thank you for pointing that. Didn't realize that writeValue method belongs to peripheral, not characteristic

Kacper20, 2016-12-22:
Hi,
Just to clarify - if you already have access to the `Characteristic` or `Descriptor` object - the most convenient is to just use method on this object.

If you have access only to connected `Peripheral` - it's nice to use convenience methods - these are making all of the calls to discover services, characteristics, descriptors so code is a way cleaner this way. You should use it when you don't have yet access to `Characteristic` on which you want to operate.

Hope it clarifies things a bit. I'm closing the issue for now - if you want to ask further questions please reopen it :)

Cheers,

closed 2016-12-22 by Kacper20