Polidea/RxBluetoothKit #87: Bluetooth is in unknown state. OSX El Capitan

vrudikov, 2016-12-31:
So i took the code from examples and created an OSX app
But i see "Bluetooth is in unknown state" for both enabled and disabled state

```
import Cocoa
import RxSwift
import RxBluetoothKit

class ViewController: NSViewController {
    
    private var isScanInProgress = false
    private var scheduler: ConcurrentDispatchQueueScheduler!
    private let manager = BluetoothManager(queue: .main)
    private var scanningDisposable: Disposable?
    fileprivate var peripheralsArray: [ScannedPeripheral] = []

    override func viewDidLoad() {
        super.viewDidLoad()

        let timerQueue = DispatchQueue(label: "com.polidea.rxbluetoothkit.timer")
        scheduler = ConcurrentDispatchQueueScheduler(queue: timerQueue)
        startScanning()
    }
    
    private func stopScanning() {
        scanningDisposable?.dispose()
        isScanInProgress = false
        print("Scanning stopped")
    }
    
    private func startScanning() {
        isScanInProgress = true
        print("Scanning...")
        scanningDisposable = manager.rx_state
            .timeout(4.0, scheduler: scheduler)
            .take(1)
            .flatMap { _ in self.manager.scanForPeripherals(withServices: nil, options: nil) }
            .subscribeOn(MainScheduler.instance)
            .subscribe(onNext: {
                self.addNewScannedPeripheral($0)
            }, onError: { error in
                print(error)
            })
    }
    
    private func addNewScannedPeripheral(_ peripheral: ScannedPeripheral) {
        let mapped = peripheralsArray.map { $0.peripheral }
        if let indx = mapped.index(of: peripheral.peripheral) {
            peripheralsArray[indx] = peripheral
        } else {
            self.peripheralsArray.append(peripheral)
        }
        print(peripheralsArray)
        DispatchQueue.main.async {
            // self.scansTableView.reloadData()
        }
    }

    override var representedObject: Any? {
        didSet {
        // Update the view, if already loaded.
        }
    }

}
```

Kacper20, 2017-01-01:
Hi there,

Basically - at the time when you're starting the app `CoreBluetooth` first state that will be emitted is `.unknown`, followed by the real value of `BluetoothState`. That's something related to `CoreBluetooth` internals.

In the code you've provided only first value is taken from the stream of `BluetoothState` values(by using `take(1)`, and it's always `unknown`. The solution is written in `README.md` of library:
```manager.rx_state
    .filter { $0 == .poweredOn }
    .take(1)
```
So to explain it: first waits for proper state, and when it's received - stream is closed by using `take(1)`.

This way it should work like a charm.
Could you try it and let us know how it goes?

Cheers,

Kacper

vrudikov, 2017-01-01:
@Kacper20 sure. i can do that.

vrudikov, 2017-01-01:
@Kacper20 works like a charm. thank you very much. closing the ticket...

closed 2017-01-01 by vrudikov