Polidea/RxBluetoothKit #92: How to access the responded data from characteristic after write .withResponse request

medAmineze, 2017-01-20:
Hi,
I recently start using this library, and it make many things much easier, thanks for this library
But now I'm having a problem and I can't deal with it
I have tried many things but without any success
So my problem is I need to write some data to characteristic and read the responded data
I can't find any way how to get the responded data and deal with it

I have something like this :

`characteristic.writeValue(data, type: .withResponse)`

and I need to get the responded data, how can I do it ?

Thanks in advance

Kacper20, 2017-01-21:
Hi there,
So, one step at a time:
When you're doing write with response it means, that your device will send you kind of `ACK` info, that purpose is to tell you that something was written to the characteristic. 

I'm just guessing, but I believe your interface looks like that: by writing to your characteristic, you're making device to start some kind of computation and return the result to you on the same characteristic.
If that's wrong - correct me, and please specify how the interface looks like from the device point of view. Sometimes it involves also setting the notification on characteristic, and then when device changes the value - read it.

If it's like I said, you first need to write value, and then there's need to read value from characteristic. It would look like:
```
characteristic.writeValue(data, type: .withResponse)
  .flatMap {.$0.readValue() }
  .subscribe(onNext...
```
The logic is:
First, you're writing the value and when you completed it with success now it's time to get the same characteristic(`$0 parameter in closure`) and readValue from characteristic.
Later, you're doing usual Rx thing, which is observing the results of your call by subscribing into `Observable`.

Let me know how it works for you.

Kacper,

medAmineze, 2017-01-29:
Hi Kacper,
Thanks for your response,
I did exactly what you told me, and it works !!
Just I changed reading the value from another characteristic
Because in my device I need to write some data to characteristic1 and then get the response from another characteristic2

My code look like this :
```

        characteristic2.writeValue(data, type: .withResponse)            
            .subscribe(onNext: { _ in
                characteristic1.readValue()
                    .subscribe(onNext: { _ in
                        let dataRecived = characteristic1.value?
                        
                     ....
                    }).addDisposableTo(disposeBag)
                
            }).addDisposableTo(disposeBag)

```

Thanks for your time Kacper.
Good work, keep going

Kacper20, 2017-01-29:
Hi,
Glad it worked.
I have one advice to you could apply to make your code more clear and in accordance with Rx conventions:

Usually, when you're trying to chain operations together(like writing, and then reading in your code) - there is `flatMap` operator that makes this kind of thing convenient to use.

You see, with your approach when you'll have 5 operations like this you will have something like "callback hell" - basically code that indents to right and is not readable.
One more problem with this approach: 
When you're receiving error in any of the operation you'll be forced to deal with it on every level(by using `subscribe(onError: { })`. 
This is how it would look like with `flatMap`.
```
        characteristic2.writeValue(data, type: .withResponse)
        .flatMap { _ in characteristic1.readValue() }
        .subscribe(onNext: { characteristic in 
            let dataReceived = characteristic.value
        }, onError: { error in 
              handle(error)
        }).addDisposableTo(disposeBag)
```
This way code is more readable, and whether error is emitted in `write`, or `read`, or maybe any operation that follows these two - it'll be always handled in one `onError` block on the end.

Hope it'll help you.
Please ask in case of any problems. 

Cheers,
Kacper

medAmineze, 2017-01-29:
Hello,

Thanks so much for helping me,
It's my first time dealing with bluetooth, and thanks to your library it make it so easy for me to deal with it.

I'll take your advice and change my code.

Thanks again,
Best regards.

closed 2017-01-29 by Kacper20