Polidea/RxBluetoothKit #93: Support macOS

cfilipov, 2017-02-17:
The bluetooth APIs on iOS and macOS are pretty much identical. The only difference is the need to import a different framework on macOS.

```Swift
#if os(OSX)
    import IOBluetooth
    #elseif os(iOS)
    import CoreBluetooth
#endif
```

Kacper20, 2017-02-17:
Hi there,

`CoreBluetooth` is supported on macOS too. In fact - our library supports macOS, we have tests for this platform and everything should work out of the box. 

`IOBluetooth` is framework exclusive to macOS which contains `CoreBluetooth` and framework for dealing with 'classic' Bluetooth. `CoreBluetooth` supports `BLE` only and this is what our library supports. Classic Bluetooth is beyond the scope of this library.

cfilipov, 2017-02-17:
Wow, indeed! I remember needing to import IOBluetooth in the past, I didn't realize CoreBluetooth existed on macOS. Thanks.

closed 2017-02-17 by cfilipov