Polidea/RxBluetoothKit #94: Periodical scanning

dneykov, 2017-02-18:
Hi everyone,
I'm trying to make periodical scan using RxBluetoothKit without success. Can anyone give some directions. I can't find something similar in the docs or sample app. What I want to do is scan for 3 seconds, pause 1 second and start scanning again for 3 seconds and so on.

Thanks.

cfilipov, 2017-02-21:
Might want to check out the code provided in [this](https://github.com/Polidea/RxBluetoothKit/issues/84#issuecomment-280563051) comment. If you remove the connection part and add a delay at the beginning you'll have what you're looking for. Note there is an issue with restarting a scan this way, but there is a workaround described there as well.

dneykov, 2017-02-21:
Hi,
the reason that I need periodical scan is that my device can start/stop advertising data at any time. Do you think that `CBCentralManagerScanOptionAllowDuplicatesKey` will do the trick. I'm open to suggestions.
Thanks.

cfilipov, 2017-02-21:
Yes, [`CBCentralManagerScanOptionAllowDuplicatesKey`](https://developer.apple.com/reference/corebluetooth/cbcentralmanagerscanoptionallowduplicateskey) sounds like it will work for you. By default scanning will only show you one advertisement per device, setting that key to `true` will ensure you get every advertisement the device sends out, so just leaving the scan going will likely suffice.

dneykov, 2017-02-21:
I just go through to your example in #84 and I think my case is similar with yours.
Thanks.

Kacper20, 2017-02-21:
@cfilipov Thanks for helping out :)

We will try to fill example app or our wiki with some more examples of advanced Rx usages.
@dneykov If you'll need solution for the problem you've mentioned earlier - 3 sec of scan, 1 sec wait over and over - please tell and we'll help you with solving this problem.

Regards,
Kacper

dneykov, 2017-02-22:
When I use `.take(1)` to get  first result scan is canceled. I wondering how can I resume scanning without creating new instance of BluetoothManager. If I call my `startScan()` function on disconnect there are 2 instances of BluetoothManager after that.
Here is my function:

```swift
private func startScanning() {
        isScanInProgress = true
        let scanOpts = [CBCentralManagerScanOptionAllowDuplicatesKey: NSNumber(value: true)];
        
        let scannedPeripherals = manager.scanForPeripherals(withServices: [Constants.UUID.UUID_SERVICE], options: scanOpts)
            .do(onSubscribe: {
                print("Searching...")
            })
            .map { (name: $0.advertisementData.localName ?? "[no name]", scannedPeripheral: $0) }
            .filter {
                ...
            }
            .take(1)
            .map { $0.scannedPeripheral.peripheral }
            .timeout(2, scheduler: MainScheduler.instance)
        
        let _ = manager.rx_state
            .filter { $0 == .poweredOn }
            .delay(0.01, scheduler: MainScheduler.instance) // <- Workaround
            .flatMap { _ in self.manager.retrieveConnectedPeripherals(withServices: [Constants.UUID.UUID_SERVICE]) }
            .flatMap { $0.isEmpty ? scannedPeripherals : Observable.from($0) }
            .flatMap { $0.connect() }
            .do(onNext: { [weak self] in
                guard let `self` = self else { return }
                self.connectionEstablish()
                self.monitorDisconnection(for: $0)
            })
            .flatMap { $0.discoverServices([Constants.UUID.UUID_SERVICE]) }
            .flatMap { Observable.from($0) }
            .flatMap { $0.discoverCharacteristics([charID])}
            .flatMap { Observable.from($0) }
            .flatMap { $0.discoverDescriptors() }
            .flatMap { Observable.from($0) }
            .flatMap { $0.readValue() }
            .retry()
            .subscribe(onNext: { descriptor in
                guard let data = descriptor.value else { return }
                print(NSString(data: data as! Data, encoding: String.Encoding.utf8.rawValue)!)
            })
            .disposed(by: self.disposeBag)
    }
```

Thanks in advance.

Kacper20, 2017-02-22:
Hello @dneykov 

`.take(1)` finishes subscription after receiving first element on stream and further cancels scanning. Is this what you want?
How it's creating two instances of `BluetoothManager`? From what I see in snippet there's only one instance of `manager`, and this should be reused.

dneykov, 2017-02-22:
Hello @Kacper20,
after first connection I call `self.monitorDisconnection(for: $0)` and when device it's disconnected I want to start scanning again. If i call `self. startScanning()` after my device is discovered twice.

dneykov, 2017-02-24:
I sill cannot make it work. Any help would be appreciated. How can I start scanning after disconnect.
Thanks.

Kacper20, 2017-02-26:
@dneykov Could you provide contents of `monitorDisconnection` function?
This will help me to know what's going on there. 
Do you call `startScanning` there?

If yes - I would try to use single disposable instead of disposeBag to ensure that there is always one stream subscribed to. This way it could make this thing with concurrent scans go away & it makes good assumptions about your intent I believe.

What do you think?

dneykov, 2017-02-26:
@Kacper20 Here is the content of `monitorDisconnection `

``` swift
    private func monitorDisconnection(for peripheral: Peripheral) {
        manager.monitorDisconnection(for: peripheral)
            .subscribe(onNext: { [weak self] (peripheral) in
                print("DISCONNECT: \(peripheral.name)")
                guard let `self` = self else { return }
                self.scannedPeripheral = nil
                self.allowOpen = false
                self.connectionLost()
                self.startScanning()
                
            })
            .disposed(by: self.disposeBag)
    }
```

How can I use single disposable? I think maybe single disposable is correct solution for me. I'm still trying to understand how exactly these Rx functions are working.

Thanks.

Kacper20, 2017-02-27:
Could you say what exactly `disposed` function is doing?
I believe it is your custom extension?

Basically:
To use single disposable I would use something which is called `SerialDisposable`. 
It looks like that:
```
let scanDisposable = SerialDisposable()
let monitorDisposable = SerialDisposable()

deinit {
    scanDisposable.dispose()
    monitorDisposable.dispose()
    // We need to remember to dispose subscriptions when our object deallocates
}
```
Then, you're assigning what you received from `subscribe` to `scanDisposable.disposable`.

It would look like:

```
monitorDisposable.disposable = manager.monitorDisconnection(for: peripheral).subscribe[...]
````
Same for scan :)

Regards,

Kacper

Kacper20, 2017-03-30:
Hello,

Any progress on this @dneykov ?

dneykov, 2017-04-07:
Hi,
I decided to implement BLE on my own.
Thanks for your efforts.

closed 2017-04-07 by dneykov