Polidea/RxBluetoothKit #95: not seing the same device list as settings app

berkankisaoglu, 2017-02-22:
Hey guys, *quick question*. 

I'm running the Example project here. 
Starting a scan. 
After couple of minutes, I only see my mac's bluetooth. 

The problem is when I go to the settings -> bluetooth and do a scan there, I was able to discover my bluetooth headphones and bluetooth speakers as well. 
Why are the lists are so different? Is that a restriction on iOS side?

Kacper20, 2017-02-22:
Hello,

This shouldn't be a case.
Are speakers and headphones BLE devices?
Are they disconnected from your Mac while performing scan while Example app?

Regards

berkankisaoglu, 2017-02-22:
My bad. So sorry to bother you. 
You were right, they were Bluetooth 3.0 devices.

Kacper20, 2017-02-22:
Not a problem :)
Have a nice day :)

closed 2017-02-22 by Kacper20