Polidea/RxBluetoothKit #96: How to continuously read a connected peripheral's RSSI?

bananapps, 2017-02-23:
Hi everyone,
I was wondering how do you continuously read the RSSI for a connected peripheral? I want to know what the updated RSSI value is at all times so I can keep track of a peripheral's signal strength.
Thank you.

bananapps, 2017-02-24:
I figured it out!
`fileprivate func triggerValueRead(for characteristic: Characteristic) {
        characteristic.setNotificationAndMonitorUpdates()
            .subscribe(onNext: {
                if self.connectedPeripheral != nil { // connectedPeripheral is the peripheral I connected to
                    self.connectedPeripheral?.readRSSI().flatMap { Observable.from($0) }
                        .subscribe(onNext: {
                            print($0.1) // rssi value
                        })
                }
       }
}
`

Kacper20, 2017-02-24:
Hello @bananapps 
That's right.
For chaining of operations you could also use `flatMap` instead of `subscribe` if you're not doing anything else.
Also, you could take peripheral from `characteristic.service.peripheral` instead.

Glad it works for you - please write in case of any questions.

Kacper

closed 2017-02-27 by Kacper20

tnmendes, 2017-04-05:
How can continuously read a peripheral's RSSI without being connected?
Or how can i continuously receive the same peripheral using this method because with this code it just received once?

```
scanningDisposable = manager.rx_state
      //.filter { $0 == .poweredOn }
      .timeout(4.0, scheduler: scheduler)
      .take(1)
      .flatMap { _ in self.manager.scanForPeripherals(withServices: nil, options:nil) }
      .subscribeOn(MainScheduler.instance)
      .subscribe(onNext: {
        self.addNewScannedPeripheral($0)
      }, onError: { error in
        print("Manager rx_state: ",error)
      })
```

moogle19, 2017-04-05:
@tnmendes 
You need to allow duplicates during scanning.
Take a look at https://developer.apple.com/reference/corebluetooth/cbcentralmanager/peripheral_scanning_options

tnmendes, 2017-04-05:
Thanks @moogle19 ...that was the solution 👍

Kacper20, 2017-04-05:
Thanks @moogle19 :)