Polidea/RxBluetoothKit #99: Make classes Open instead of Public (Swift 3)

jack4git, 2017-03-03:
I'm wanting to create a subclass of ScannedPeripheral or Peripheral (to add a property). In Swift 3, I can't just do "class pairingPeripheral: ScannedPeripheral { ..." -- that throws and error "Cannot inherit from non-open class ScannedPeripheral outside its defining module" ...

Changing Public to Open in the various classes would fix ...

Kacper20, 2017-03-05:
Hello @jack4git 

What is your case for inheriting? Why don't you just compose this additional property and `ScannedPeripheral` in new object?
Just trying to find out, maybe it makes sense :)

Regards,

Kacper

jack4git, 2017-03-14:
Hey Kacper -- apologies for delay in responding. 

My workaround was composing as you suggested. 

So my question is, are you intentionally keeping access to Public to prevent subclassing because it's a bad idea? Or is this just a new Swift 3 feature that you aren't yet taking advantage of?

Kacper20, 2017-03-15:
Hello,

It's not our intent to use this Swift 3 feature, because in our opinion subclassing in this case is unncessary. We've designed classes as closed components that shouldn't be modified(at least we don't see any profit from this).
In fact I think that it would be a good idea to mark these classes as `final` and we will do it in next version of library.

jack4git, 2017-03-15:
Thanks for quick response --

closed 2017-03-15 by jack4git