Polidea/RxBluetoothKit #100: Estimote catch instance uuid

tnmendes, 2017-03-21:
I am using estimote proximity beacon and i am trying to obtain the advertising UUID and i am having some difficulty.
So write now i have this code:

```
print("add : advertisementData : ", peripheral.advertisementData.advertisementData)

print("add : serviceUUIDs : ", peripheral.advertisementData.serviceUUIDs?[0].description ?? "zero")
```

that the result is:

```
add : just UUID :  AF471412-5AC5-47AA-AFEE-6B9E5495190F
add : advertisementData :  ["kCBAdvDataIsConnectable": 0, "kCBAdvDataServiceData": {
    FEAA = <00e1edd1 ebeac04e 5defa017 24267fd3 25bf0000>;
}, "kCBAdvDataServiceUUIDs": <__NSArrayM 0x170057a00>(
FEAA
)
]

add : serviceUUIDs :  FEAA

```

where the "edd1 ebeac04e 5defa017" is namespace and "24267fd3 25bf" is the instance.
how can i catch the instance ( "24267fd3 25bf") in code?

Kacper20, 2017-04-26:
Hello @tnmendes 

Sorry for the delay. If I understand you correctly you want to retrieve service data from advertisement.
You should retrieve whole data by using `CBAdvertisementDataServiceDataKey`
This looks something like:
```swift
let data = advertisementData[CBAdvertisementDataServiceDataKey][CBUUID(string: "FEAA")]
```
`Data` is collection beginning from Swift 3 so you should be able to easily retrieve wanted bytes from it.

Please write in case of any further questions.

Regards,

Kacper

Kacper20, 2017-05-04:
Closing it for now - please feel free to reopen in case of any further concerns.

Regards,

Kacper

closed 2017-05-04 by Kacper20