Polidea/RxBluetoothKit #101: how can i rxbluetoothkit in background mode?

choiks14, 2017-03-29:
how can i rxbluetoothkit in background mode?

please help me..

please give me some example code.

thanks

Kacper20, 2017-03-29:
Hello.

The one think you should do is to add background bluetooth mode to Info.plist.

Then, when requested for example characteristic monitoring - you should get callback while your app is in background and be able to make some light operations for limited time.

choiks14, 2017-03-29:
No.
listenOnRestoredState is not working..

this is my code 
1.scan
2.listenOnRestoredState
3.connect a device.
4.kill signal (``` kill(getpid(), SIGKILL);```)
5.Debug->Attach ...
6.off ble and on ble
7.below code is run..
but listenOnRestoredState callback is not working.
```
 func application(_ application: UIApplication, didFinishLaunchingWithOptions launchOptions: [UIApplicationLaunchOptionsKey: Any]?) -> Bool {
```

```
manager.listenOnRestoredState()
            .subscribe(onNext: { (restoredState) in
                print("test")
            },onError: {error in
                print(error)
            },onCompleted:{
                print("asdf")
            },onDisposed:{
                 print("monitorDisconnection onDisposed" )
            }).disposed(by: self.disposeBag)
```

Kacper20, 2017-03-30:
Hello @choiks14 

Could you show me how are you creating your `BluetoothManager` and show me more of your code?

Regards,

Kacper

choiks14, 2017-03-31:
```
class MainViewController: UIViewController {
class BleService{
    static let shared:BleService = BleService()
    
    
    let manager = BluetoothManager(queue:.main,options: [CBCentralManagerOptionRestoreIdentifierKey:"SMARTPLUS" as AnyObject])
    //let manager = BluetoothManager(queue:DispatchQueue.global(qos: .background),options: [CBCentralManagerOptionRestoreIdentifierKey:"SMARTPLUS" as AnyObject])
}
```

MainViewController.swift
```
class MainViewController: UIViewController {
    override func viewDidLoad() {
        super.viewDidLoad()
        
       
        self.listen()
        
  
    }

    func listen(){
        manager.listenOnRestoredState()
            .subscribe(onNext: { (restoredState) in
                print("test")
            },onError: {error in
                print(error)
            },onCompleted:{
                print("asdf")
            },onDisposed:{
                print("monitorDisconnection onDisposed" )
            }).disposed(by: self.disposeBag)
    }
}
```

listen not working..

choiks14, 2017-03-31:
please help me.........

Kacper20, 2017-03-31:
Hello @choiks14 

Please check documentation: https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1

Especially section on recreating your `BluetoothManager`. You should do this and call `listenOnRestoredState` if you have identifiers in `launchOptions` `UIApplicationDelegate` method.

Please write if you need further help

choiks14, 2017-03-31:
no.. 

native CBCentralManager is called.

but rxbluetoothkit is not called

```
 func centralManager(_ central: CBCentralManager, willRestoreState dict: [String : Any]) {
   //is called by system..
}
```

```
 manager.listenOnRestoredState()
            .subscribe(onNext: { (restoredState) in
                // is never called.
            },onError: {error in
                print(error)
            },onCompleted:{
                print("asdf")
            },onDisposed:{
                print("monitorDisconnection onDisposed" )
            }).disposed(by: self.disposeBag)
```

Kacper20, 2017-03-31:
Hello,

How do you know whether `willRestore` state from `CoreBluetooth` is called? Are you setting breakpoints in `RxCBCentralManager` or maybe in some other way?

Could you set breakpoints in this file:
https://github.com/Polidea/RxBluetoothKit/blob/master/Source/RxCBCentralManager.swift
In lines 60 and 93 and check which one is called first? 

Regards,

choiks14, 2017-03-31:
RxCBCentralManager is not called.

My code is another project.

1.scan
2.connect
3.kill
4.attach
5. willRestoreState  was called by system

Kacper20, 2017-03-31:
Are you using code in exactly the same places?(I mean code that is creating managers and listening on restored state).
Our lib is basically wrapper on `CoreBluetooth` stuff, so this should work as intended when used in appropriate way.

choiks14, 2017-03-31:
Could you give me a listenOnRestoredState sample code?

choiks14, 2017-03-31:
I woke up AppDelegate.

but below code not called.

```
_ = self.manager.listenOnRestoredState()
            .subscribe(onNext: { (restoredState) in
                print(restoredState.bluetoothManager)
                print("test")
            },onError: {error in
                print(error)
            },onCompleted:{
                print("asdf")
            },onDisposed:{
                print("monitorDisconnection onDisposed" )
            })
```

Kacper20, 2017-03-31:
Could you show me whole project or send me some sample project in which this issue happens? 

What is `launchOptions[UIApplicationLaunchOptionsKey.bluetoothCentrals]` returning?

choiks14, 2017-03-31:
https://github.com/choiks14/Test

this is my code

1.build
2.scan
3.kill
4.debug attach
5.some ble change

not working....

please help me..

Kacper20, 2017-03-31:
Hello,

As I said - have you read the documentation I sent to you?
You have a function `initListen` in your VC. You're calling it in `viewDidLoad` of your `UIViewController` which is probably too late.

The things you need to do:
1. Place the code that reads identifiers of restored `CBCentralManager` objects from `launchOptions` in your `AppDelegate`(now it's empty).
2. Check there whether this list contents your `"SMARTPLUS"` identifier
3. If yes - recreate your `BluetoothManager` and start subscribing to `listenOnRestoredState` method.

Could you please follow these three steps?

Henawey, 2017-04-09:
I don't know if you mean killing the app by double click on home button and swipe up the app from background application (Closed by the user), If this what you mean, the CoreBluetooth Framework will never begin the app in background (Works only with "iBeacon").

It only should works if your app in the background and the iOS kill it for some reason (memory, etc...) then when you register for scanning and you work with central background mode. It'll work.

if you need to simulate to run the app in the background, you should work with fetch background mode and edit your schema to launch the app in background fetch.

Kacper20, 2017-04-26:
Hello @choiks14,

Any progress on this?

rdgier, 2017-05-15:
I am having the same issue. What I do currently:  
1. Start app  
2. When view has loaded, start scanning  
3. When app is about to be sent to the background/terminated, I subscribe to listenOnRestoredState()

I tried subscribing to listenOnRestoredState in the app delegate, but my view is called before applicationDidFinishLaunching.  
Also tried subscribing to listenOnRestoredState() before subscribing to rx_state to start scanning. Again no scanning while the app was in the background.

I have no idea what to try anymore. If anyone has some suggestions, then that would be great.

rdgier, 2017-05-18:
Little update from my side. Appearently there is bluetooth scanning happening in the background from an earlier version of my app, but it doesn't show in the debug cpu process at all.  
This is expected behaviour from Apple if I read their docs correctly, since my btManager it's lifecycle is the same as the app's lifecycle.  

(Don't know if I need to create a seperate issue for this)  
I have one question for listenOnRestoreState though, how do I debug this? I do not know when the subscribe method is being called and I do not see any of the onNext etc. print messages I placed in the console.

HAKASHUN, 2017-09-02:
I'm having the same issue.
I tried subscribing to `listenOnRestoredState` as soon as possible in AppDelegate, but the subscriber couldn't always get `.Next` because [`willRestoreStateSubject`](https://github.com/Polidea/RxBluetoothKit/blob/2915b388da2c75f9acc4f178d136b4b2b3fe27c7/Source/RxCBCentralManager.swift#L56) emit `.Next` event before `listenOnRestoredState` is subscribed.

I resolved this issue by applying patches in `RxCBCentralManager.swift`.

1. Changed `InternalDelegate.willRestoreStateSubject` to `BehaviorSubject` from `PublishSubject`
2. Edited `RxCBCentralManager.rx_willRestoreState` for the change.

```
class RxCBCentralManager: RxCentralManagerType {

    ...

    @objc private class InternalDelegate: NSObject, CBCentralManagerDelegate {

        ...

        let willRestoreStateSubject = BehaviorSubject<[String: AnyObject]?>(value: nil)
        
        ...

    }

    ...

    /// Observable which infroms when central manager is about to restore its state
    var rx_willRestoreState: Observable<[String: AnyObject]> {
        return internalDelegate.willRestoreStateSubject.flatMap { state -> Observable<[String: AnyObject]> in
          guard let state = state else { return Observable.empty() }
          return Observable.just(state)
        }
    }

    ....

}
```

Kacper20, 2017-09-06:
Hi @HAKASHUN,

You're right, many thanks! Could you make a pull request to the library with your changes?
If not - I will reimplement it right away and release a new version.

Best Regards,

Kacper

HAKASHUN, 2017-09-07:
Thanks for your reply!

I'll make a pull request as soon as possible.

Kacper20, 2017-10-02:
I think that should resolve all of the race conditions in our code. Thanks @HAKASHUN! :)

closed 2017-10-02 by Kacper20