Polidea/RxBluetoothKit #102: Some Questions about the Example App

nanoq, 2017-03-29:
I’m new in Bluetooth (RxSwift) and I’ve some questions regarding your example App:
1) In the function startScanning() is a scan timeout attached. But I can’t find any user information about the issue: ‚This scan is timed out!‘
	What is the best solution to implement this functionality?
2) In the case the user has the bluetooth powered off and started the App, there is no functionality to inform the user about this issue, yet.
	How can I implement this functionality?

Kacper20, 2017-03-29:
1. It depends on how the app is implemented. If you're using RxSwift `.timeout()` operator you one choice:
`timeout()` is returning some internal RxSwift error if none of the items was emitted in specified time.  You could then detect this error and react to it in `subscribe(onError: { }` block.

2. You could check `rx_state` and subscribe to it - when `poweredOff` is received it means that bluetooth is turned off.

Kacper20, 2017-03-31:
@nanoq Any progress on this?

nanoq, 2017-03-31:
Yes! I'm stepped onward.
Everything will be okay in the end. If it's not okay, it's not the end. ;-)

Kacper20, 2017-03-31:
Cool! 👍 Let me know if you have further questions, I'll be happy to help!

Kacper20, 2017-04-26:
Closing issue for now - please open another issue in case of any further questions.

Regards,
Kacper

closed 2017-04-26 by Kacper20