Polidea/RxBluetoothKit #103: Unable to cancel connection

thierrybucco, 2017-03-29:
Hello,
I am unable to cancel a connection.
Here is my code:

```
peripheral.connect()
            .flatMap { $0.discoverServices([configServiceUUID]) }
            .flatMap { Observable.from($0) }
            .flatMap { $0.discoverCharacteristics([configCharacteristicUUID])}
            .flatMap { Observable.from($0) }
            .flatMap { $0.readValue() }
            .subscribe(onNext: { [ weak self ] in
                let data = $0.value
                print(String(data: data!, encoding: .utf8))
                peripheral.cancelConnection() 
            })
```

It's pretty straightforward but peripheral is still connected.

Do you have an idea why?

Thanks

Kacper20, 2017-03-30:
Hello,

So basically the reason behind it is that `cancelConnection` returns `Observable`. Until you subscribe to this returned `Observable` - nothing is really done. 
Once you'll do it - connection will be canceled.

Kacper20, 2017-04-02:
Hello @thierrybucco

thierrybucco, 2017-04-05:
It works. Thanks a lot.

closed 2017-04-05 by thierrybucco