Polidea/RxBluetoothKit #104: monitorWrite() bug?

jackliang086, 2017-03-30:
when i subscribe monitorWrite() ,everytime i get Characteristic.value is always nil. why? thanks

Kacper20, 2017-03-30:
What happened when you're doing 2 things instead:
Subscribe `monitorWrite` and once that fires `Next` trigger `read` on Characteristic. This should do what you intended.

Regards,
Kacper

jackliang086, 2017-03-30:
there is the code snippet:

characteristic.monitorWrite()
            .subscribe(onNext: {[weak self] (c:Characteristic) in
                
                print(c.value) // <- the c.value is always nil

            }).addDisposableTo(disposeBag)

my intention is that i wanted to get the feedback after i write value to a characteristic

thanks!!!

Kacper20, 2017-03-30:
@jackliang086 
Exactly, I suggest to do another operation in sequence:
```
characteristic.monitorWrite()
.flatMap { $0.readValue }
.subscribe(onNext: {
    print($0.value)
}.addDisposableTo(disposeBag)
```

Please let me know how it works for you :)

jackliang086, 2017-03-30:
let me try,wait a seconds^_^

jackliang086, 2017-03-30:
it works , but the read value is not what i writed T.T

Kacper20, 2017-03-30:
That's weird. Are you writing this value from outside, or within your app from another place?

jackliang086, 2017-03-30:
i just write the value one place ,
actually,i found this problem even if i only use corebluetooth framework. the value from the write value callback function is different to what i writing

Kacper20, 2017-03-30:
It may be a problem with your firmware internals if that's the case.

jackliang086, 2017-03-30:
yeah , i have the same idea

jackliang086, 2017-03-30:
but i test it by the android platform,it is perfect.-_-

Kacper20, 2017-03-30:
Are you using notifications or indications on Android? How does it work there exactly? I could ask some Android friends that are taking care of https://github.com/Polidea/RxAndroidBle library and maybe we'll come up with something.

Regards,

jackliang086, 2017-03-30:
yes. but i do not think is the key place. my intention is easy, just test the callback value from the writed callback function on android platfrom. and everything is prefect.do you think there are some bugs on corebluetooth framework.because your library is base on corebluetooth framework

Kacper20, 2017-03-30:
Could you tell me exactly what API and function calls are you using on Android?

jackliang086, 2017-03-30:
public void onCharacteristicWrite(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic, int status)

jackliang086, 2017-03-30:
thanks a lot for your patience ^_^

Kacper20, 2017-03-31:
@jackliang086 Could you tell me how are you writing value? I suspect that you're using `withoutResponse`. Could you try using `withResponse` and check the results?

jackliang086, 2017-04-01:
no, i am using withResponse

c!.writeValue(data, type: CBCharacteristicWriteType.withResponse)

Kacper20, 2017-04-26:
Hello @jackliang086 

Any updates on this?

Kacper20, 2017-08-01:
I'm closing this as of now. If you have any other questions - please feel free to reopen issue, or open another one.

Kacper

closed 2017-08-01 by Kacper20