Polidea/RxBluetoothKit #106: MTU exchange requests?

thierrybucco, 2017-04-19:
Hello,

I would like to increase MTU size, do you know how to do this?
I am unable to find some documentation about this.

Thanks a lot.

Thierry

Cierpliwy, 2017-04-20:
Hello,

As I know iOS is automatically exchanging MTU and only API related to it is maximumWriteValueLength(for:) which allows you get negotiated value.

More details can be found on SO:
http://stackoverflow.com/questions/41977767/iosble-get-negotiate-mtu

thierrybucco, 2017-04-20:
Thanks a lot.
I have a question. For sending saying 2KB of data should I segment them into 158 bytes chunks myself and do multiple write with response (for resend if it failed) or is it managed by CoreBluetooth?

closed 2017-05-03 by thierrybucco