Polidea/RxBluetoothKit #107: onCompleted never called when reading a characteristic value

thierrybucco, 2017-04-24:
Hi,

When i read a characteristic value I never get onCompleted, do you know why?
Thanks a lot.

```
tileConnect.peripheral.connect()
	.flatMap { $0.discoverServices([ self.tileConnectServices.tc1ServiceUUID, self.tileConnectServices.wifiServiceUUID ]) }
	.flatMap { Observable.from($0) }
	.flatMap { $0.discoverCharacteristics([ self.tileConnectServices.wifiCharacteristicUUID ])}
	.flatMap { Observable.from($0) }
	.flatMap { $0.readValue() }
	.subscribe(onNext: {
		print("onNext")
	}, onError: { error in
		print("onError: \(error.localizedDescription)")
	}, onCompleted: {
		print("onCompleted") // <-- Never called
	}, onDisposed: { [ weak self ] in
		print("onDisposed")
	})

```

closed 2017-04-25 by thierrybucco

Kacper20, 2017-04-25:
Hello @thierrybucco 

Sorry for the delay.
Have you managed to resolve it somehow? Please write in case you need any help.

Regards,

Kacper

thierrybucco, 2017-04-27:
Hi @Kacper20, yes I found the problem related to #108.
Thanks for your help.