Polidea/RxBluetoothKit #108: Unable to readValue

thierrybucco, 2017-04-25:
Hi,

I have a method which returns an Observable.  (next event is coming from RxBluetooth readValue), I don't know why I am unable to receive next event.

onDisposed is called when I dispose subscriber.

Do you know what I am doing wrong?

Thanks

```
func getLinkStatus(for tileConnect: TileConnect) -> Observable<( (String, String), (String, String) )?> {

	return Observable.create { [ weak self ] observer in

		// it works when I simulate received data
		// observer.onNext( ( ( "192.168.250.1", "38.996" ), ( "8.8.8.8", "192.064" ) ) ) 

		if let strongSelf = self {

			tileConnect.peripheral.connect()
			.flatMap { $0.discoverServices([ strongSelf.tileConnectServices.wifiServiceUUID ]) }
			.flatMap { Observable.from($0) }
			.flatMap { $0.discoverCharacteristics([ strongSelf.tileConnectServices.networkConnectivityCheckUUID ])}
			.flatMap { Observable.from($0) }
			.flatMap { $0.readValue() }
			.subscribe(onNext: {
				if let data = $0.value {
					if let delimitedString = String(data: data, encoding: .utf8) {
						let valuesArr = delimitedString.components(separatedBy: ";")
						if valuesArr.count == 4 {
							observer.onNext( ( ( valuesArr[0], valuesArr[1] ), ( valuesArr[2], valuesArr[3] ) ) )
						}
					}
				}
			}, onError: { error in
				LOG_SENSOR_MANAGER("getLinkStatus. onError: \(error.localizedDescription)")
			}, onCompleted: {
				LOG_SENSOR_MANAGER("getLinkStatus. onCompleted")
			}, onDisposed: {
				LOG_SENSOR_MANAGER("getLinkStatus. onDisposed")
			})
			.disposed(by: strongSelf.disposeBag)
		}

		return Disposables.create()
	}
}
```

Kacper20, 2017-04-25:
Hello @thierrybucco 

Are you able to discover service and characteristic? Is read even called? 
Also - it's a good practice to return actual disposable from `.create`. method, so you should be returning instance you receive from `subscribe` method. 
Is `Observable.create` necessary here, or maybe you're doing this only to illustrate something?

Regards,

Kacper

thierrybucco, 2017-04-25:
Hello @Kacper20,

how to know id read is called?
I added .debug() and I got:

```
2017-04-25 17:29:44.639: SensorsManager.swift:477 (getLinkStatus(for:)) -> subscribed
2017-04-25 17:29:44.642: SensorsManager.swift:475 (getLinkStatus(for:)) -> subscribed
2017-04-25 17:29:44.644: SensorsManager.swift:473 (getLinkStatus(for:)) -> subscribed
2017-04-25 17:29:44.653: SensorsManager.swift:471 (getLinkStatus(for:)) -> subscribed
2017-04-25 17:29:44.658: SensorsManager.swift:469 (getLinkStatus(for:)) -> subscribed
2017-04-25 17:29:44.662: SensorsManager.swift:467 (getLinkStatus(for:)) -> subscribed
2017-04-25 17:29:44.666: SensorsManager.swift:467 (getLinkStatus(for:)) -> Event next(RxBluetoothKit.Peripheral)
2017-04-25 17:29:44.670: SensorsManager.swift:467 (getLinkStatus(for:)) -> Event completed
2017-04-25 17:29:44.670: SensorsManager.swift:467 (getLinkStatus(for:)) -> isDisposed
```

When I don't return an Observable but call RxBluetooth directly in method it works.
For Observable.create I am not sure, I read an exemple on how to create an Observable. What do you suggest?

Thanks a lot.

thierrybucco, 2017-04-25:
@Kacper20, about what you said: "Also - it's a good practice to return actual disposable from .create. method, so you should be returning instance you receive from subscribe method." 

Could you give me an exemple?

Thanks

Kacper20, 2017-04-25:
Right now you're creating one level too much of indirection.
Your interface states that method returns `Observable`. Content of this method should just be:
```
return tileConnect.peripheral.connect()
			.flatMap { $0.discoverServices([ strongSelf.tileConnectServices.wifiServiceUUID ]) }
			.flatMap { Observable.from($0) }
			.flatMap { $0.discoverCharacteristics([ strongSelf.tileConnectServices.networkConnectivityCheckUUID ])}
			.flatMap { Observable.from($0) }
			.flatMap { $0.readValue() }
```
This way you're doing subscribe when you're calling `getLinkStatus` method. 
`Observable.create` is really useful when you're doing things like wrapping callback-based API or something like that. Right here you're just operating on `Observable` instances so this should be fine.

Could you tell me where debug was added? Also - please check how it works without wrapping it in `Observable.create`. It's very important to check when `disposeBag` is being reinitialised - this could be helpful for further investigation.

thierrybucco, 2017-04-26:
Hi, thanks for your answer. Here is where debug has been added.

```
.debug()
.flatMap { $0.discoverServices([ strongSelf.tileConnectServices.wifiServiceUUID ]) }
.debug()
.flatMap { Observable.from($0) }
.debug()
.flatMap { $0.discoverCharacteristics([ strongSelf.tileConnectServices.networkConnectivityCheckUUID ])}
.debug()
.flatMap { Observable.from($0) }
.debug()
.flatMap { $0.readValue() }
.debug()
.subscribe(onNext: {
...
```

I understand about Observable.create, but what I want to achieve is getting data from bluetooth peripheral then send a corresponding object to observers, is there another way to do this instead of using Oberservable.create?

Kacper20, 2017-04-26:
@thierrybucco do you have multiple observers? Could you show how are you using `getLinkStatus` function? Is it used in multiple places at the same time? Using `Observable.create` in this way is the same in result as using it like I showed before.

thierrybucco, 2017-04-26:
@Kacper20 there is 1 observer. I changed my code to as you suggested.

```
func getLinkStatus(for tileConnect: TileConnect) -> Observable<(Characteristic)> {
		return tileConnect.peripheral.connect()
			.flatMap { $0.discoverServices([ self.tileConnectServices.wifiServiceUUID ]) }
			.flatMap { Observable.from($0) }
			.flatMap { $0.discoverCharacteristics([ self.tileConnectServices.networkConnectivityCheckUUID ])}
			.flatMap { Observable.from($0) }
			.debug()
			.flatMap { $0.readValue() }
	}
```

Then getLinkStatus is called like that:

```
private func startMonitoringLinkStatus() {
		monitoringLinkStatusDisposable = sensorsManager.getLinkStatus(for: tileConnect).asObservable()
			.debug()
			.subscribe(onNext: { [ weak self ] data in
				if let data = data.value {
					if let delimitedString = String(data: data, encoding: .utf8) {
						let valuesArr = delimitedString.components(separatedBy: ";")
						if valuesArr.count == 4 {
							self?.formBuilder.linkStatus = ( ( valuesArr[0], valuesArr[1] ), ( valuesArr[2], valuesArr[3] ) )
						}
					}
				}
			})
	}

private func stopMonitoringLinkStatus() {
	monitoringLinkStatusDisposable?.dispose()
}
```

Here is logs I get:

```
2017-04-26 10:00:46.325: SensorsAssistantListSensorDetailsViewModel.swift:132 (startMonitoringLinkStatus()) -> subscribed
2017-04-26 10:00:46.327: SensorsManager.swift:464 (getLinkStatus(for:)) -> subscribed
```

thierrybucco, 2017-04-27:
I spent 3 days trying to fix this problem.
Here is a summary of what I found:

- I was able to read value from a characteristic, but not from another one. (LightBlue is able to get value)
- After a moment I got "Unexpectedly disconnected" "Read Permission Denied" always for the same characteristic.

So I had the genius idea to simply reboot my iPad and all is working like a charm... 

Thanks you all for you help.

Thierry

closed 2017-04-27 by thierrybucco