Polidea/RxBluetoothKit #110: Correct way to for stopping being notified

thierrybucco, 2017-05-03:
Hello,

I am wondering what is the best way for stopping being notified.
Unfortunately I am unable to stop correctly. I never receive onCompleted.
Here is my code:

```

	func startWifiNetworksScanning(for tileConnect: TileConnectBLEDevice) -> Observable<WifiNetworkEntry?> {

		return Observable.create { [ weak self ] observer in
			if let strongSelf = self {
				 tileConnect.peripheral.characteristic(with: TileConnectBLEDeviceCharacteristic.wifiAccessPointsList)
					.flatMap { $0.setNotificationAndMonitorUpdates() }
					.takeWhile({ characteristic in
						return characteristic.isNotifying == true
					})
					.subscribe(onNext: {
						if let data = $0.value {
							if let wifiAccessPointsDelimitedString = String(data: data, encoding: .utf8) {
								let wifiNetworkEntry = WifiNetworkEntry(from: wifiAccessPointsDelimitedString)
								observer.onNext(wifiNetworkEntry)
							}
						}
					}, onError: { error in
						LOG_SENSOR_MANAGER("getWifiNetworks. onError: \(error.localizedDescription)")
						observer.onError(error)
					}, onCompleted: { // <---- Never goes here
						LOG_SENSOR_MANAGER("getWifiNetworks. onCompleted")
						observer.onCompleted()
					}, onDisposed: {
						LOG_SENSOR_MANAGER("getWifiNetworks. onDisposed")
					})
				.disposed(by: strongSelf.disposeBag)
			}
			return Disposables.create()
		}
	}

	func stopWifiNetworksScanning(for tileConnect: TileConnectBLEDevice) {
		tileConnect.peripheral.characteristic(with: TileConnectBLEDeviceCharacteristic.wifiAccessPointsList)
			.flatMap { $0.setNotifyValue(false)}
			.subscribe(onNext: { _ in
				print("setNotifyValue(false).")
			})
	}
```

Do you have some hints?

Thanks a lot.

Thierry

thierrybucco, 2017-05-03:
Hello,

I got it working like this:

```
func startWifiNetworksScanning(for tileConnect: TileConnectBLEDevice) -> Observable<WifiNetworkEntry?> {

		return Observable.create { [ weak self ] observer in
			if let strongSelf = self {
				strongSelf.scanningWifiNetworksDisposable = tileConnect.peripheral.characteristic(with: TileConnectBLEDeviceCharacteristic.wifiAccessPointsList)
					.flatMap { $0.setNotificationAndMonitorUpdates() }
					.subscribe(onNext: {
						if let data = $0.value {
							if let wifiAccessPointsDelimitedString = String(data: data, encoding: .utf8) {
								let wifiNetworkEntry = WifiNetworkEntry(from: wifiAccessPointsDelimitedString)
								observer.onNext(wifiNetworkEntry)
							}
						}
					}, onError: { error in
						LOG_SENSOR_MANAGER("getWifiNetworks. onError: \(error.localizedDescription)")
						observer.onError(error)
					}, onCompleted: {
						LOG_SENSOR_MANAGER("getWifiNetworks. onCompleted")
						observer.onCompleted()
					}, onDisposed: {
						LOG_SENSOR_MANAGER("getWifiNetworks. onDisposed")
						observer.onCompleted()
					})
			}

			return Disposables.create()
		}
	}

	func stopWifiNetworksScanning(for tileConnect: TileConnectBLEDevice) {
		scanningWifiNetworksDisposable?.dispose()
	}
```

Is it correct?

Thanks a lot.

Thierry

Kacper20, 2017-05-04:
Hello @thierrybucco 
One thing to change is to not embed everything in `Observable.create` call. This does nothing useful in this case and you could simplify your code by a lot.
About ending subscription to `Observable` instance:
Object which is taking care of managing subscription is `Disposable` which is returned from the call to `subscribe`. By calling `dispose` on it you're telling that you are not longer interested in subscription and want it to stop. By calling it in this specific case we're under the hood doing disabling of notifications on characteristic.

So in summary:
Answer you've provided is the correct one 👍 

Best Regards,

Kacper

thierrybucco, 2017-05-04:
Hi @Kacper20,

thanks for your comment. 
I understand about not embedding everything in an Observable, but In this case, this method get data from BLE device and create an object (WifiNetworkEntry).

I don't want caller of startWifiNetworksScanning() to have to deal with raw device data coming from device.

Do you have a suggestion?

Thanks!

Kacper20, 2017-05-04:
I would do it like that(preserving the behaviour you have currently):

```swift
func startWifiNetworksScanning(for tileConnect: TileConnectBLEDevice) -> Observable<WifiNetworkEntry?> {
    return tileConnect.peripheral.characteristic(with: TileConnectBLEDeviceCharacteristic.wifiAccessPointsList)
        .flatMap { $0.setNotificationAndMonitorUpdates() }
        .takeWhile({ characteristic in
        return characteristic.isNotifying == true
    })
    .flatMap { (characteristic) -> Observable<WifiNetworkEntry>? in
        if let data = $0.value, let wifiAccessPointsDelimitedString = String(data: data, encoding: .utf8) {
            return WifiNetworkEntry(from: wifiAccessPointsDelimitedString)
        } else {
            return .empty()
        }
    }
}
```

Then - you could subscribe to `Observable` instance returned from this method and when subscription is no longer valid - call `dispose` on returned `Disposable` instance.

Hope it helps,

Kacper

thierrybucco, 2017-05-04:
Of course!
Thanks a lot for your help and your work.

Thierry

closed 2017-05-04 by thierrybucco

thierrybucco, 2017-05-04:
I got: 

.flatMap { (characteristic) -> Observable<WifiNetworkEntry>? in <<<< Cannot convert value of type '(_) -> Observable<WifiNetworkEntry>?' to expected argument type '(Characteristic) -> _'

Kacper20, 2017-05-04:
Uhh, sorry.

Edited my answer to correct version - was writing this code on GitHub without compiling :) 
Now this should work well

thierrybucco, 2017-05-04:
Also should it be:

`return Observable.just(WifiNetworkEntry(from: wifiAccessPointsDelimitedString))` instead of `return WifiNetworkEntry(from: wifiAccessPointsDelimitedString)` ?

Kacper20, 2017-05-04:
Yep, of course @thierrybucco :)