Polidea/RxBluetoothKit #119: flatMap writeValue for characteristic do not return proper Observable<Characteristic>

danielbotka, 2017-06-05:
I'm using RxBluetoothKit and cannot figure out the following issue. I'm having this wrapper function to write a characteristics (I'm trying to write many of them with concat, it works nice):

```
func writeCharacteristicData(for peripheral: Peripheral, characteristicIdentifier: CharacteristicIdentifier, value: Data?) -> Observable<Characteristic> {
guard let value = value else {
  return Observable.error(BTError.noDataProvided(description: "Missing value for: \(characteristicIdentifier)"))
}
return peripheral
  .connect()
  .flatMap { $0.writeValue(value,
    for: characteristicIdentifier,
    type: .withResponse) }
}
```

According to RxBluetoothKit documentation `.flatMap { $0.writeValue }` is one of the easiest way to write in a characteristic. Actually it works just fine with `readValue` and even with this `writeValue`, the write is successful, and if I read it back with `readValue` after that, the `characteristic.value` is OK in oNext:.

My problem is that this function returns an `Observable<Characteristic>` and my assumption was that it will give me back the written value. But that's not the case, or at least it does not work for me. It gives me back previous values or nil, I could not figure this out so far what's the logic on the emitted `Observable<Characteristic>`. It would be really nice if I can get back the written value. What am I doing wrong?

I even tried the longer ways, and this one, mentioned on github:

```
characteristic.writeValue(data, type: .withResponse)
.subscribe { event in
    //respond to errors / successful read
}
```
The emitted characteristic (in the `onNext`: callback) is not the one I wrote to the device. It's a previous one or nil. The funny thing is if I try to read it back, and write after that, the `characteristic.value` will be the one that I've read with `$0.readValue`. But if I recompile, write a completely new one, it's still the same I've read before. It's like RxBluetoothKit cached a value or I have no idea whatsoever.

Can this be a problem with the ble device? Maybe it's a thread issue or something like that?

I'd really appreciate any kind of advice or hints.

I checked the example project and works the same, after writing it emits the last read value, not the last written value.

Thanks,

Daniel

Cierpliwy, 2017-06-05:
This is normal on iOS. I didn't find any section in Bluetooth specification which is talking about this behaviour (I'll definitely search for it), but I assume that device may not update characteristic's value which user sent. Therefore Characteristic object contains latest value which was read. In case value wan't read it's `null`.

danielbotka, 2017-06-05:
Oh, so it's expected behaviour. I debugged my code for like 3 hours because I was pretty sure that doing something wrong. 

Anyways, I think it's kind of weird API design to give back a characteristic observable after successful write. I would assume that it is the written value, not something that was read sometime ago, or even nil. But I get the point, I rewrite these parts of my code.

May I close this ticket?

Cierpliwy, 2017-06-05:
Sure. We won't change this behaviour probably, because it was designed in that way. However I'll check what BLE specification is saying about that.

closed 2017-06-06 by danielbotka