Polidea/RxBluetoothKit #123: Xcode 9 beta 3 errors

huguesbr, 2017-07-24:
RxBluetoothKit was working great with `Xcode 9 beta 2` but after updating to `beta 3`, I got the following errors:

```
.../RxBluetoothKit/Source/Observable+QueueSubscribeOn.swift:78:7: Type 'QueueSubscribeOn<Element>' does not conform to protocol 'ObservableType'
```

```
.../RxBluetoothKit/Source/Observable+QueueSubscribeOn.swift:132:70: Cannot convert value of type 'QueueSubscribeOn<Element>' to expected argument type '(Event<_>) -> Void'
```

```
.../RxBluetoothKit/Source/Observable+QueueSubscribeOn.swift:164:76: Cannot convert return expression of type 'Observable<QueueSubscribeOn<Self.E>.E>' to return type 'Observable<Self.E>'
```

Any ideas?

Kacper20, 2017-07-24:
Hi @huguesbr 

We have already PR for this - I will merge it in two hours and release next version.

Thanks for the patience,

Kacper

Kacper20, 2017-07-24:
Hi,

It's already released as `3.0.14` version. 

Thanks again!

If you feel your issue has been resolved - please close the issue :)

huguesbr, 2017-07-24:
yep problem solved, thx! 👍

closed 2017-07-24 by huguesbr