Polidea/RxBluetoothKit #125: retrieveConnectedPeripherals is not working for orphan connections. Does it need options?

RamblinWreck77, 2017-07-26:
Hello, I'm fighting an issue currently involving "orphan" connections, where the app has disconnected however iOS is maintaining the physical link indefinitely until the peripheral power is disconnected. This has the effect of not only making the peripheral unable to be connected to again by the app, but also blocking all other devices from seeing it when scanning.

While trying to fix the underlying issue I updated my scan code to check if one of our devices was already connected before beginning a scan. The issue I'm having is retrieveConnectedPeripherals is returning no devices even when in this "orphan" state. The  Observable<[Peripheral]> array is always []. 

Looking at the CoreBluetooth method:
`centralManager.retrieveConnectedPeripherals(withServices: serviceUUIDs).map(RxCBPeripheral.init))
`

The documentation says:
"The list of connected peripherals can include those that are connected by other apps and that will need to be connected locally using the connect(_:options:) method before they can be used."

Yet I don't see a way to pass those options in, which seems important if the method is supposed to return a peripherial. It's possible this is not related to why it's not working, but I thought I would ask. 

Thanks in advance for any help you can offer!

Kacper20, 2017-07-27:
Hi,
Thanks for asking!
`options` is an optional parameter to `connect` function. The full interface looks as follows:
```
public func connect(options: [String: AnyObject]? = nil) -> Observable<Peripheral> {
    return manager.connect(self, options: options)
}
```
So you could definitely pass some options, but none of the known to me should affect `retrieveConnectedPeripherals` result call. What's your expected behavior? Do you want to disconnect all of the already connected devices? I think that we could help you with the problem but we need a bit more context :)

Regards,

Kacper

RamblinWreck77, 2017-07-27:
Thanks for the quick reply, feel free to close this. I believe we've isolated the issue to the peripheral's bluetooth module locking up somehow. 

Basically for one part of our transfer process a lot of async requests can get batched up as observables. Let's say there's 50 requests in this queue, and #5 experiences an unexpected disconnect. Right now 6-50 are then immediately "collapsing" and throwing:

`[CoreBluetooth] API MISUSE: <CBPeripheral: ***, identifier = CB274A23-1C22-4206-90FC-925E9DBF3DDF, name = "", state = disconnected> can only accept commands while in the connected state` 

-One for each of the queued up commands. 

This ends up locking the peripheral's Bluetooth module somehow, as it is now undetectable by the main device or any other (Android or Apple), even leaving the building and coming back (guaranteeing that the physical link is broken) does not recover the sensor. The only solution to this is a power cycle... so we're working on improving our code to prevent this scenario. 

Is there some kind of RxBluetooth feature to hard-block outgoing commands if the state is disconnected?

Kacper20, 2017-07-27:
Hi, @RamblinWreck77 

Actually, it is. Every time when the write is invoked there is a code that is checking the connection to the peripheral. What kind of commands are you using? I will make sure that there is check everywhere, to have more control over that.

From what I see, there are not any checks for reads and I believe there should be one. I will release a new version tomorrow with this issue fixed. 

Regards,

RamblinWreck77, 2017-07-27:
So the logs when this happens look like this: (removed some identifying info)

`
🔥 SENSOR ERR READ: {RxBluetoothError}: Peripheral Disconnected Error: The connection has timed out unexpectedly.
🔸 error_readItems_catchall: {RxBluetoothError}: Peripheral Disconnected Error: The connection has timed out unexpectedly.
🔥 SENSOR ERR WRITE: {RxBluetoothError}: Peripheral Disconnected Error: -
🔥 SENSOR ERR READ: {RxBluetoothError}: Peripheral Disconnected Error: -
[CoreBluetooth] API MISUSE: <CBPeripheral: 0x1742f3200, identifier = CB274A23-1C22-4206-90FC-925E9DBF3DDF, name = "", state = disconnected> can only accept commands while in the connected state
`

Where there's about ~30-50 sets of the write/read/[CoreBluetooth] logss

Where the write log fires from a do-on-error from:
`
self.writeCharacteristic.writeValue(data, type: self.writeType)
`

and read:

`
self.readValue(terminateCharacter: terminateCharacter)
`

which is:

`
return readCharacteristic.monitorValueUpdate()
            .map { $0.value ?? Data() }
            .takeUntilInclusive { data in data.contains(terminateCharacter) }
            .reduce(Data()) { combinedData, result in combinedData.appending(result) }
`

RamblinWreck77, 2017-07-27:
(edited reply)

RamblinWreck77, 2017-07-30:
@Kacper20  Hey man, any update on a patch for this? Hope you're having a great weekend!

Kacper20, 2017-07-30:
Hi there, @RamblinWreck77. Thanks for asking, I've had small delay with this.
I've just submitted #126 
This should be merged tomorrow and throw errors in `Observable` stream when `read` is called while `Peripheral` is in `disconnected` state. 

Have a great day,

Kacper

RamblinWreck77, 2017-07-30:
Great news! Appreciate it!

Kacper20, 2017-07-31:
Hi @RamblinWreck77,

We're working on this, but some problems evolved. We want to be sure that when peripheral is disconnected read on peripheral is never called internally. With the solution we have already in code error is thrown but it's done later than call to read. To solve it properly and never call `read` on peripheral when it's disconnected we need one more day of designing code.

Regards,
Kacper

RamblinWreck77, 2017-07-31:
@Kacper20 thanks for the update!

Kacper20, 2017-08-01:
Hi @RamblinWreck77,

Could you please check library version from master to see whether your issue is fixed? If it is - I will release new version today.

Regards,
Kacper

RamblinWreck77, 2017-08-01:
@Kacper20 Sorry for the late update, it's been crazy today. 

I verified that I pulled from master / my local code includes the commits that had the fixes. The issue is still occurring. Once the Peripherial is disconnected the "WRITE" commands are blocked but "READ" commands each are followed by the [CoreBluetooth] error.

I did notice however that this time the logs are saying the invalid state is "disconnecting" not "disconnected", is is possible that "disconnect-ing" isn't being filtered properly? 

Here what I get:
`
🔥 SENSOR ERR WRITE: {RxBluetoothError}: Peripheral Disconnected Error: -
23:01:41.122 [INFO] 	
🔥 SENSOR ERR READ: {RxBluetoothError}: Peripheral Disconnected Error: -
2017-08-01 19:01:41.122295-0400 [-] [CoreBluetooth] API MISUSE: <CBPeripheral: 0x1702fc700, identifier = CB274A23-1C22-4206-90FC-925E9DBF3DDF, name = "", state = disconnecting> can only accept commands while in the connected state
`

RamblinWreck77, 2017-08-01:
@Kacper20 

Unfortunately I just did some power-off/interrupt testing (unplugging power mid-communication to see how it's handled) and was able to get it to attempt to read when in the "disconnected" state too.
`
🔥 SENSOR ERR WRITE: {RxBluetoothError}: Peripheral Disconnected Error: -
`
`
🔥 SENSOR ERR READ: {RxBluetoothError}: Peripheral Disconnected Error: -
`
`
[CoreBluetooth] API MISUSE: <CBPeripheral: 0x1700fcf80, identifier = CB274A23-1C22-4206-90FC-925E9DBF3DDF, name = "", state = disconnected> can only accept commands while in the connected state
`

Kacper20, 2017-08-02:
@RamblinWreck77 Thanks for checking this! We're gonna work on this on our side today. That's unexpected, because code shouldn't allow it. I'm gonna reproduce it and let you know more about this.

Kacper20, 2017-08-04:
Hi @RamblinWreck77,

We've found the possible issue and added tests for this behavior. They're passing and I believe everything should be fine now. Could you confirm, please?

Regards,

Kacper

RamblinWreck77, 2017-08-04:
@Kacper20  I'm out of the office through most of the weekend, I'll try and sneak away to test this tomorrow. Thanks again for knocking this out

Kacper20, 2017-08-05:
@RamblinWreck77 No rush, next time you have the opportunity is fine :-)

Thanks!

RamblinWreck77, 2017-08-05:
@Kacper20  Looks like that got it! No more [CoreBluetooth] errors in my logs during disconnect and I can't get it to jam/lock the peripheral up anymore. 

Thanks!

RamblinWreck77, 2017-08-07:
Closing this, because the current branch on master appears to solve the problem. Thanks again! looking forward to the next release @Kacper20

closed 2017-08-07 by RamblinWreck77

ddwang, 2018-06-25:
Guys, quick question: we are seeing something similar and found this thread.  Just to confirm, @RamblinWreck77 are you saying by allowing a read command to a disconnected CBPeripheral object locked up your (physical) peripheral device?

pejn, 2019-01-03:
@ddwang Were you able to resolve this issue?
I am having this warning message as well. It causes an issue of not being able to connect afterwards.
"API MISUSE: <CBPeripheral: (...) state = disconnecting> can only accept commands while in the connected state"