Polidea/RxBluetoothKit #130: Issue writing and reading chunked

MarvinNazari, 2017-08-09:
Hi, Ive been trying to write (chunked) and read large values, found some stuff in the issues but still no luck
here is what Im doing now, it never reaches to `step2` of handshake
Thanks in advance, love the project btw.

```swift
override func viewDidLoad() {
    super.viewDidLoad()

    manager.rx_state
        .filter { $0 == .poweredOn }
        .take(1)
        .flatMap { _ in self.manager.scanForPeripherals(withServices: nil) }
        .filter { $0.peripheral.name == "TT 3015" }.take(1)
        .flatMap { $0.peripheral.connect() }
        .flatMap { MTKHandshakeProcessor(peripheral: $0).process(libCommand: self.libCommand) }
        .subscribe(onNext: { response in
            print("NEXT -> \(response.data.utf8String ??? "")")
        }).addDisposableTo(disposeBag)
}

enum MTKHandshake {
    case step1
    case step2
    
    func payload(libCommand: LibCommandGateway) -> Data {
        switch self {
        case .step1: return libCommand.dataCommand(2, command: "REQV")
        case .step2: return libCommand.dataCommand(2, command: "0 8 1.1 IOS")
        }
    }
}

struct MTKHandshakeProcessor {
    let peripheral: Peripheral
    
    init(peripheral: Peripheral) {
        self.peripheral = peripheral
    }
    
    func process(libCommand: LibCommandGateway) -> Observable<CommandResult> {
        let processor = MTKResponseProcessor()
        let handler = { (data: Data) in
            processor.findCommandResponse(data: data, libCommand: libCommand)
        }
        
        return peripheral
            .writeAndRead(data: MTKHandshake.step1.payload(libCommand: libCommand), untilFound: handler)
            .flatMap { _ in self.peripheral.writeAndRead(data: MTKHandshake.step2.payload(libCommand: libCommand), untilFound: handler) }
    }
    
}

struct CommandResult {
    let data: Data
}

extension Peripheral {

    func writeChunked(writer: Characteristic, data: Data) -> Observable<Characteristic> {
        let chunks = data.chunked(byMaxSize: maximumWriteValueLength(for: .withResponse))
        return Observable.from(chunks).flatMap { writer.writeValue($0, type: .withResponse) }
    }
    
    func writeAndRead(data dataToSend: Data, untilFound: @escaping ((Data) -> BinaryCommand?)) -> Observable<CommandResult> {
        
        let pendingData = NSMutableData()
        
        // This observable will take care of accumulating data received from notifications
        let result = characteristic(with: MTKDeviceCharacteristic.read)
            .flatMap { $0.monitorValueUpdate() }
            .takeWhile { characteristic in
                return untilFound(pendingData as Data) != nil
            }
            .reduce(pendingData, accumulator: { (data, characteristic) -> NSMutableData in
                if let packetData = characteristic.value {
                    data.append(packetData)
                }
                return data
            })
        
        
        let writes = characteristic(with: MTKDeviceCharacteristic.write)
            .flatMap { self.writeChunked(writer: $0, data: dataToSend) }
        
        return Observable.zip(result, writes, resultSelector: { (result: NSMutableData, query: Characteristic) -> CommandResult in
            return CommandResult(data: result as Data)
        })
    }
}

extension Data {
    func chunked(byMaxSize: Int) -> [Data] {
        return (self as NSData).dataChunks(ofSize: UInt(byMaxSize))
    }
}

```

Kacper20, 2017-08-09:
Hi,

Could you please let me know where exactly code is not working and what is your intended behavior(with some details on the order of operations)?
It’ll be easier for me to help you having some more context

Regards,

Kacper

MarvinNazari, 2017-08-09:
Hi, basically i don't get any thing on `MTKDeviceCharacteristic.read` so seems like `monitorValueUpdate ` dosen't work.

this is the log

```
[RxBLEKit|DEBG|11:14:59.449]: CentralManager(6243726848) didDiscover(peripheral: Peripheral(uuid: 378CA308-D985-426E-A346-03AEC00A60A8, name: Optional("Tinitell 3015")), rssi: -37)
[RxBLEKit|DEBG|11:14:59.756]: CentralManager(6243726848) didConnect(to: Peripheral(uuid: 378CA308-D985-426E-A346-03AEC00A60A8, name: Optional("Tinitell 3015")))
[RxBLEKit|DEBG|11:15:00.125]: Peripheral(uuid: 378CA308-D985-426E-A346-03AEC00A60A8, name: Optional("Tinitell 3015")) didDiscoverServices(services: Optional("[Service(uuid: 000018A0-0000-1000-8000-00805F9B34FB, id: 6245731136)]"), error: nil)
[RxBLEKit|DEBG|11:15:00.185]: Peripheral(uuid: 378CA308-D985-426E-A346-03AEC00A60A8, name: Optional("Tinitell 3015")) didDiscoverCharacteristicsFor(for:Service(uuid: 000018A0-0000-1000-8000-00805F9B34FB, id: 6245731136), characteristics: Optional("[Characteristic(uuid: 00002AA0-0000-1000-8000-00805F9B34FB, id: 6241882240)]"), error: nil)
[RxBLEKit|DEBG|11:15:11.539]: Peripheral(uuid: 378CA308-D985-426E-A346-03AEC00A60A8, name: Optional("Tinitell 3015")) didDiscoverCharacteristicsFor(for:Service(uuid: 000018A0-0000-1000-8000-00805F9B34FB, id: 6245731136), characteristics: Optional("[Characteristic(uuid: 00002AA0-0000-1000-8000-00805F9B34FB, id: 6241882240), Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6241873216)]"), error: nil)
[RxBLEKit|DEBG|11:15:11.703]: Peripheral(uuid: 378CA308-D985-426E-A346-03AEC00A60A8, name: Optional("Tinitell 3015")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6241873216), value: nil, error: nil)
```
as you see it writes the data but i don't receive anything back

MarvinNazari, 2017-08-09:
I found the issue, i didn't start the reading

```swift
let writesAndStartTheReading = characteristic(with: MTKDeviceCharacteristic.write)
    .flatMap { self.writeChunked(writer: $0, data: dataToSend) }
    .flatMap { _ in self.characteristic(with: MTKDeviceCharacteristic.read) }
    .flatMap { self.readValue(for: $0) }
```

MarvinNazari, 2017-08-09:
What my flow is/need to be:
1. I write some data to write characteristic
2. Listen to read characteristic
3. readValue until i get a special cursor in the data
4. Done.

(in writing i need to split them into chunks as well)

Kacper20, 2017-08-21:
Hello @MarvinNazari,

Sorry, I was on vacations. I will take a look tomorrow at your issue.

Thanks for patience,

Kacper

MarvinNazari, 2017-08-21:
Hi @Kacper20, No worry

This is my current flow:
https://gist.github.com/MarvinNazari/4d5ea9b906d2cffa15b4810e0bd65721

I needed to fork to add a new func to Peripheral which reads the data without listening [here](https://gist.github.com/MarvinNazari/4d5ea9b906d2cffa15b4810e0bd65721#file-more-swift)

MarvinNazari, 2017-08-21:
There is something wrong here this is what I'm doing right now and the logs are really wrong, it seems like that `writeValue` dosen't wait for response so it just continues and do all the write all att once

```swift
extension Peripheral {
    
    func writeChunked(data: Data) -> Observable<Characteristic>{
        return characteristic(with: MTKDeviceCharacteristic.write)
            .flatMap { self.writeChunked(data: data, writer: $0) }
    }
    
    func writeChunked(data: Data, writer: Characteristic) -> Observable<Characteristic> {
        let chunks = data.chunked(byMaxSize: 150)
        
        return Observable.from(chunks)
            .flatMap { chunk in
                writer.writeValue(chunk, type: .withResponse).debug("----------- WRITER ------------")
            }.debug("WRITING")
    }
}
```

you see that `didWriteValueFor` callback gets called after `WRITING` is disposed.

```
2017-08-21 15:53:16.821: ----------- WRITER ------------ -> subscribed
2017-08-21 15:53:16.822: ----------- WRITER ------------ -> subscribed
2017-08-21 15:53:16.822: ----------- WRITER ------------ -> subscribed
2017-08-21 15:53:16.823: ----------- WRITER ------------ -> subscribed
2017-08-21 15:53:16.823: ----------- WRITER ------------ -> subscribed
2017-08-21 15:53:16.824: ----------- WRITER ------------ -> Event completed
2017-08-21 15:53:16.824: ----------- WRITER ------------ -> isDisposed
2017-08-21 15:53:16.824: WRITING -> Event completed
2017-08-21 15:53:16.824: WRITING -> isDisposed
[RxBLEKit|DEBG|13:53:16.824]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
2017-08-21 15:53:16.824: ----------- WRITER ------------ -> Event next(RxBluetoothKit.Characteristic)
2017-08-21 15:53:16.824: WRITING -> Event next(RxBluetoothKit.Characteristic)
2017-08-21 15:53:16.827: WRITING -> subscribed
2017-08-21 15:53:16.828: ----------- WRITER ------------ -> subscribed
2017-08-21 15:53:16.829: ----------- WRITER ------------ -> Event completed
2017-08-21 15:53:16.829: ----------- WRITER ------------ -> isDisposed
2017-08-21 15:53:21.715: ----------- WRITER ------------ -> Event next(RxBluetoothKit.Characteristic)
2017-08-21 15:53:21.715: WRITING -> Event next(RxBluetoothKit.Characteristic)
2017-08-21 15:53:21.716: ----------- WRITER ------------ -> Event completed
2017-08-21 15:53:21.717: ----------- WRITER ------------ -> isDisposed
2017-08-21 15:53:21.717: WRITING -> Event completed
2017-08-21 15:53:21.717: WRITING -> isDisposed
[RxBLEKit|DEBG|13:53:21.822]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
[RxBLEKit|DEBG|13:53:21.822]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
[RxBLEKit|DEBG|13:53:21.822]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
[RxBLEKit|DEBG|13:53:21.823]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
[RxBLEKit|DEBG|13:53:21.823]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
[RxBLEKit|DEBG|13:53:21.823]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6243949344), value: nil, error: nil)
```

MarvinNazari, 2017-08-23:
@Kacper20 any news?

Kacper20, 2017-08-23:
Hello @MarvinNazari,

Firstly I'm looking into your last post. Let's clarify this and go with further problems then :)
The problem is that what you're doing is actually writing all of the things at once.

`Observable.from(chunks)` emits all of the chunks at once. It later calls all of the writes, too.
What you should be doing is to use `concat` operator in ReactiveX. This will allow you to perform operations sequentially.

If something is not clear - please ask. I will answer your further questions as fast as I can :)

MarvinNazari, 2017-08-23:
@Kacper20 How would `concat` look like with a progress subject?
because I'm writing a big data I need some progress for it

```swift
class MTKPeripheralWriterss {
    private let peripheral: Peripheral, size: Int

    private let progress = Variable<Float>(0)
    
    var progeessObservable: Observable<Float> {
        return progress.asObservable()
    }
    
    init(peripheral: Peripheral, size: Int) {
        self.peripheral = peripheral
        self.size = size
    }
    
    func writeChunked(data: Data, writer: Characteristic) -> Observable<Characteristic> {
        let chunks = data.chunked(byMaxSize: size)
        return Observable<Data>.from(chunks)
            .flatMapWithIndex { chunk, idx -> Observable<(Data, Int)> in
                return .just((chunk, idx))
            }
            .flatMap { chunk, idx in Observable.zip(writer.writeValue(chunk, type: .withResponse), Observable<Int>.just(idx)) }
            .do(onNext: { chunk, idx in
                print("----------------------- PROGRESS \(Float(idx + 1) / Float(chunks.count))")
            })
            .flatMap { _, _ in Observable<Characteristic>.just(writer) }
    }
}
```

Kacper20, 2017-08-23:
Hi,

Suppose you have a function like that, which provides the actual data.

```swift
    private func data() -> [(String, Int)] {
        return [
                ("a", 1),
                ("b", 2),
                ("c", 3),
                ("d", 4)
            ]
    }
```

Then, you're something like that with concat(delay is for the purpose of showing that operations are serial:

```swift
        let result = Observable.from(data()).map { Observable.just($0).delay(5.0, scheduler: MainScheduler.instance).debug("LOG", trimOutput: false) }
        result.concat().subscribe(onNext: { value in
            print(value)
        }).disposed(by: bag)
```

Logs are:
```
2017-08-23 12:36:42.809: LOG -> Event next(("a", 1))
("a", 1)
2017-08-23 12:36:47.733: LOG -> Event completed
2017-08-23 12:36:47.733: LOG -> isDisposed
2017-08-23 12:36:47.733: LOG -> subscribed
2017-08-23 12:36:52.733: LOG -> Event next(("b", 2))
("b", 2)
2017-08-23 12:36:52.734: LOG -> Event completed
2017-08-23 12:36:52.734: LOG -> isDisposed
2017-08-23 12:36:52.734: LOG -> subscribed
2017-08-23 12:36:57.736: LOG -> Event next(("c", 3))
("c", 3)
2017-08-23 12:36:57.737: LOG -> Event completed
2017-08-23 12:36:57.737: LOG -> isDisposed
2017-08-23 12:36:57.737: LOG -> subscribed
2017-08-23 12:37:02.740: LOG -> Event next(("d", 4))
("d", 4)
2017-08-23 12:37:02.741: LOG -> Event completed
2017-08-23 12:37:02.741: LOG -> isDisposed
```

Which as you see is sequential - subscribe on observable is called after the earlier one completed. You could handle progress in your `subscribe(onNext:)` method handling. 

Kacper

MarvinNazari, 2017-08-23:
Hi @Kacper20,

This is what I do here (which look like you example) but it seems that it stills sends everything at once.
```swift
return Observable.from(chunks)
            .map { writer.writeValue($0, type: .withResponse) }
            .concat()
```

Kacper20, 2017-08-23:
Hm, could you please add rx debug call to `writer.writeValue($0, type: .withResponse` and send me results?

MarvinNazari, 2017-08-23:
```
2017-08-23 14:10:13.638: Start------- -> subscribed
number of chunks that will be sending 1
2017-08-23 14:10:13.653: --- WRITING CHUNK -> subscribed
[RxBLEKit|DEBG|12:10:13.783]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6246021344), value: nil, error: nil)
2017-08-23 14:10:13.784: --- WRITING CHUNK -> Event next(RxBluetoothKit.Characteristic)
2017-08-23 14:10:13.784: Start------- -> Event next(RxBluetoothKit.Characteristic)
number of chunks that will be sending 1790
2017-08-23 14:10:13.804: --- WRITING CHUNK -> subscribed
2017-08-23 14:10:13.873: --- WRITING CHUNK -> Event completed
2017-08-23 14:10:13.873: --- WRITING CHUNK -> isDisposed
2017-08-23 14:10:13.873: Start------- -> Event completed
2017-08-23 14:10:13.873: Start------- -> isDisposed
[RxBLEKit|DEBG|12:10:13.873]: Peripheral(uuid: 0F9667AB-60B2-4BCB-8D76-49E92945FCFC, name: Optional("Tinitell 6105")) didWriteValueFor(for:Characteristic(uuid: 00002AA1-0000-1000-8000-00805F9B34FB, id: 6246021344), value: nil, error: nil)
2017-08-23 14:10:13.873: --- WRITING CHUNK -> Event next(RxBluetoothKit.Characteristic)
2017-08-23 14:10:13.874: --- WRITING CHUNK -> isDisposed
```

MarvinNazari, 2017-08-23:
The issue I'm seeing right now is that `handle` in this class dosen't wait for `peripheralGateway.write` to get completed so it goes directly to `self.peripheralGateway.read `
```swift
class MTKSoftwareInstallerGateway {
    fileprivate let peripheral: Peripheral
    fileprivate let libCommand: LibCommandGateway
    fileprivate let processor = MTKResponseProcessor()
    fileprivate let peripheralGateway: PeripheralGateway
    
    //NOTE: The maximum value from peripheral dosent work for this, this is hardcoded
    fileprivate let maxChunkSize = 0x1400
    
    init(peripheral: Peripheral, libCommand: LibCommandGateway = MTKLibCommandGateway()) {
        self.peripheral = peripheral
        self.libCommand = libCommand
        peripheralGateway = PeripheralGateway(peripheral: peripheral)
    }

    func handle(firmwarePath: String) -> Observable<FotaInstallStatus> {
        guard let firmware = firmware(from: firmwarePath) else { return .error(TinitellError.failedDownloadingUpdate()) }
        
        let findCommandResponse = { (data: Data) in
            self.processor.findCommandResponse(data: data, libCommand: self.libCommand)
        }
        
        return peripheralGateway
            
            .write(data: BTFotaCommands.start(chunks: chunked(firmware: firmware).count).data(using: libCommand)).debug("Start-------", trimOutput: false)
            .flatMap { _ in self.peripheralGateway.write(data: BTFotaCommands.commandsByChunks(chunks: self.chunked(firmware: firmware), libCommand: self.libCommand)) }
            .flatMap { _ in self.peripheralGateway.write(data: BTFotaCommands.end.data(using: self.libCommand)) }
            .flatMap { _ in
                self.peripheralGateway.read(untilFound: findCommandResponse).timeout(25, scheduler: MainScheduler.instance).retry(2)
            }
            .flatMap { binaryCommand -> Observable<BinaryCommand> in
                precondition(binaryCommand.data.utf8String == "fota")
                return self.peripheralGateway.read(untilFound: self.processor.findFotaResponse)
            }.debug("FIRST FOTA RESPONSE")
            .flatMap { binaryCommand -> Observable<FotaInstallStatus> in
                guard let string = binaryCommand.data.utf8String else { preconditionFailure("Not  a string \(binaryCommand)") }
                guard let int = Int(string) else { preconditionFailure("Not an int \(string)") }
                guard let result = FotaInstallStatus(rawValue: int) else { preconditionFailure("Not in range \(int)") }
                guard result == .transferSuccess else { throw NSError.cancelledError() }
                return .just(result)
            }
            //FIXME: Should read for FotaInstallStatus.updateSuccess
    }
}
```

MarvinNazari, 2017-08-24:
@Kacper20 found the issue it was from my side, I needed to add `takeLast(1)` in my writer, but it would be nice to have a helper function in the library that would write long data.

closed 2017-08-24 by MarvinNazari

Kacper20, 2017-08-24:
Hi @MarvinNazari,
Glad it worked. We'll think about it, but probably because the problem is kind of complex(there are different ways to write values to peripheral). We'll definitely write some examples to do so in documentation

avivm, 2019-06-18:
In case anyone come here for a working solution.
Chunking the data:

`extension Data {
    func chunked(byMaxSize: Int) -> [Data] {
        let length = self.count
        var offset = 0
        var chunks: [Data] = []
        
        repeat {
            // Get the length of the chunk
            let thisChunkSize = ((length - offset) > byMaxSize) ? byMaxSize : (length - offset);
            
            // Get the chunk
            let chunk = self.subdata(in: offset..<offset + thisChunkSize )
            chunks.append(chunk)
            
            // Update the offset
            offset += thisChunkSize
            
        } while (offset < length)
        
        return chunks
    }
}`

Writing the data:
`       let chunks = out.chunked(byMaxSize: size)
        Observable.from(chunks)
            .map { self.writeCharacteristic!.writeValue($0, type: .withResponse).asObservable() }
            .concat()
            .subscribe(
                onNext: { (dataModel) in },
                onError: { [weak self] (error) in
                },
                onCompleted: { [weak self] in
                },
                onDisposed: {})
            .disposed(by: disposeBag)`

Hope it helps.

avivm, 2019-06-18:
In case anyone come here for a working solution.
Chunking the data:
```
extension Data {
    
    func chunked(byMaxSize: Int) -> [Data] {
        let length = self.count
        var offset = 0
        var chunks: [Data] = []
        
        repeat {
            // Get the length of the chunk
            let thisChunkSize = ((length - offset) > byMaxSize) ? byMaxSize : (length - offset);
            
            // Get the chunk
            let chunk = self.subdata(in: offset..<offset + thisChunkSize )
            chunks.append(chunk)
            
            // Update the offset
            offset += thisChunkSize
            
        } while (offset < length)
        
        return chunks
    }
    
}
```
Writing the chunks:
```
let chunks = data.chunked(byMaxSize: 23)
Observable.from(chunks)
    .map { self.writeCharacteristic!.writeValue($0, type: .withResponse).asObservable() }
    .concat()
    .subscribe(
        onNext: { (dataModel) in },
        onError: { [weak self] (error) in
        },
        onCompleted: { [weak self] in
        },
        onDisposed: {})
    .disposed(by: disposeBag)
```
Hope it helps!